espressif/esp-rainmaker #2: Question:Rainmaker support

abhatikar, 2020-04-18:
Will we have Rainmaker support for the Esp32 Wroom or Huzzah32 ?

shahpiyushv, 2020-04-25:
We will start giving out support for ESP32 on-demand and later open up to a wider audience. Can you send the email id you used for registering on ESP RainMaker to esp-rainmaker-admin@espressif.com so that it can be enabled for you?

DanielJimenezHernandez, 2020-05-29:
same issue here i want to start testing in the platform with the devkit-c that i have can this also be enabled for me?

Di-Strix, 2020-05-30:
Not so long ago, I tested Rainmaker again on esp-wroom-32 and after the update it started working fine. But it needs to setup CLI first

shahpiyushv, 2020-06-01:
@DanielJimenezHernandez , the ESP32 support is now enabled by default for all. As mentioned by  @Di-Strix , you need to set-up the CLI and use the claim command to get the credentials.

shahpiyushv, 2020-10-12:
Closing this issue as it is no more relevant.

closed 2020-10-12 by shahpiyushv