espressif/esp-rainmaker #3: esp_rmaker_core error

AleConti, 2020-04-23:
Hi, I build and flash firmware to an ESP32WROOM, and get this error:

E (824) esp_rmaker_core: Failed to initialise Node Id
E (824) app_main: Could not initialise ESP RainMaker. Aborting!!!

This is Switch example, it build OK and then flash OK too. BOOT and Led output are ok too, it turn ON and OFF the output led. But WiFi dont start .

Thanks!!!
Ale Conti

scamberlin, 2020-04-24:
I tried with this type of problem on an m5stack. I think the support for esp32 card is incomplete ? There are problems with retrieving the node id on esp32 (it's work with esp32s2) and with claiming support ?

shahpiyushv, 2020-04-24:
We will start giving out support for ESP32 on-demand and later open up to a wider audience. Can you send the email id you used for registering on ESP RainMaker to esp-rainmaker-admin@espressif.com?

AleConti, 2020-04-25:
> I tried with this type of problem on an m5stack. I think the support for esp32 card is incomplete ? There are problems with retrieving the node id on esp32 (it's work with esp32s2) and with claiming support ?

Yes, sure, I advanced one more step, and now it creates NODE ID , but fails when claim MQTT Credentials... 
I (851) wifi:Init dynamic rx buffer num: 32
I (881) esp_rmaker_core: Node ID ----- 30AEA40CA360
E (881) esp_rmaker_core: Failed to initialise MQTT Config. Credentials missing.
E (881) app_main: Could not initialise ESP RainMaker. Aborting!!!

scamberlin, 2020-04-25:
The code actually generate certificates for esp32s2 but not for esp32. It seem to be in progress :)

AleConti, 2020-04-25:
Yes, self claiming service is only for esp32s2 now, we must wait for our esp32 ;) ...

scamberlin, 2020-04-25:
Or we can provide some help... This project is great !

AleConti, 2020-04-25:
Yes of course! In the menuconfig of ESP32s2 the self claim option appears to setup, where the URL and the Host MQTT endpoint are configured to generate the MQTT certificate. Very interesting project, I have a batch of esp32WROOM to apply it, and the application is very interesting as presented. I'm working with AWS free trier for the backend, and AWS amplify with React Native for the App, but this project directly saves me a lot of time and cost.

closed 2020-06-01 by AleConti