espressif/esp-rainmaker #5: Could not build the project with error

jamesjian007, 2020-05-15:
Hi
error: 

esp-rainmaker/components/led_strip/src/led_strip_rmt_ws2812.c:155:14: error: implicit declaration of function 'rmt_get_counter_clock'; did you mean 'rmt_get_source_clk'? [-Werror=implicit-function-declaration]
  STRIP_CHECK(rmt_get_counter_clock((rmt_channel_t)config->dev, &counter_clk_hz) == ESP_OK,
              ^~~~~~~~~~~~~~~~~~~~~

shahpiyushv, 2020-05-15:
@jamesjian007 , what is your esp-idf version? Are you using the master branch?

jamesjian007, 2020-05-16:
@shahpiyushv I am using 4.0.

shahpiyushv, 2020-05-16:
@jamesjian007, please switch to the master branch and try again.

jamesjian007, 2020-05-17:
Fixed by use the esp-idf-tools-setup-2.3.exe tool and reinstall with the Master branch. thanks

closed 2020-05-17 by jamesjian007