espressif/esp-rainmaker #7: Push Button UI element

NemesisXB, 2020-05-25:
This is a feature request to add a push button UI element. Basically like a doorbell. A button that does not remember state. It can possible send true when you push it, and when you push it again, it will send true as the value again.
It can be used for example to trigger an open or close of a gate or garage door.

mrutyunjay-patel, 2020-06-18:
👍  My vote for this feature !
also add if possible with modes  Push-to-ON and Push-OFF , Time Duration

daniel-frenkel, 2021-02-23:
I agree, this is the most basic UI element and is very much needed. Please add. Thank you

aca18mj, 2021-04-03:
Is there a way to add custom UI elements or are we stuck with only 3 (text, switch, slider)?

shahpiyushv, 2021-04-05:
@aca18mj we will keep on adding new UI elements (though I admit that there aren't many options as of now), but if by "custom" you mean something that the app itself does not support by default, then you can make the changes in the source code itself and rebuild the app. Ofcourse, we would try to add more and more options ourselves after we flush out the current tasks in the pipeline. You may also file PRs if you have created something yourselves. We added a hue slider some time back via [this PR](https://github.com/espressif/esp-rainmaker-android/pull/1).

CodorStelian, 2021-10-05:
The push button ui element is already available in android app, but I don't know if it is available in iOS app too.
Use this in your custom param:
`esp_rmaker_param_add_ui_type(param, esp.ui.trigger);`

NemesisXB, 2021-10-05:
> esp.ui.trigger

Any reason this is not documented?
https://rainmaker.espressif.com/docs/standard-types.html

xhivo97, 2021-11-16:
> 
> 
> > esp.ui.trigger
> 
> Any reason this is not documented? https://rainmaker.espressif.com/docs/standard-types.html

I noticed that as well, it is mentioned here:
https://docs.espressif.com/projects/esp-rainmaker/en/latest/c-api-reference/rainmaker_standard_types.html

shahpiyushv, 2021-11-17:
@NemesisXB @xhivo97 this wasn't documented as it wasn't yet added in iOS. There are other some undocumented types as well which we will add soon. Sorry for the delay on our part.

xhivo97, 2021-11-22:
> @NemesisXB @xhivo97 this wasn't documented as it wasn't yet added in iOS. There are other some undocumented types as well which we will add soon. Sorry for the delay on our part.

Thanks for the reply.
I don't have an iOS device to test right now, will using `esp.ui.trigger` break anything for iOS users?

me-chiel, 2022-08-16:
a pushbutton would be very helpful! 
I hope it will be available for IOS as well.

xhivo97, 2022-08-17:
> a pushbutton would be very helpful! I hope it will be available for IOS as well.

It works on iOS but the icons are not fully working. As a side note, the app on android looks more polished IMO and some icons are missing, e.g. on Android a device named `Garage` will actually show a garage door icon while on iOS it's a generic IC icon.

Looks like the push button is getting fixed in future iOS releases https://github.com/espressif/esp-rainmaker-ios/issues/12#issuecomment-1202040451

Michiel-FMI-ImProvia, 2022-08-17:
> It works on iOS but the icons are not fully working. As a side note, the app on android looks more polished IMO and some icons are missing, e.g. on Android a device named `Garage` will actually show a garage door icon while on iOS it's a generic IC icon.

I need to experiment with this then. Thanks for the heads-up! What I need is  the esp.ui.trigger then?
Usually IOS app's take some more trouble to being released in the app, and Android is pretty flexible with updates. 

> Looks like the push button is getting fixed in future iOS releases [espressif/esp-rainmaker-ios#12 (comment)] (https://github.com/espressif/esp-rainmaker-ios/issues/12#issuecomment-1202040451)

I love this app sofar, can't wait to get everything working as I want it.

xhivo97, 2022-08-17:
> I need to experiment with this then. Thanks for the heads-up! What I need is the esp.ui.trigger then? Usually IOS app's take some more trouble to being released in the app, and Android is pretty flexible with updates.

Yes, it's `esp.ui.trigger` you can also use the `ESP_RMAKER_UI_TRIGGER` macro.

> I love this app sofar, can't wait to get everything working as I want it.

I've been using it for 9 months for my garage door opener, it's pretty solid.

Michiel-FMI-ImProvia, 2022-08-17:
I'll have a look later. Thanks for the input. 

Did you get the OTA update working via Arduino by any chance?

me-chiel, 2022-08-17:
Thanks,

I got it working now!

shahpiyushv, 2023-02-22:
esp.ui.trigger is now supported in iOS too. Closing this issue.

closed 2023-02-22 by shahpiyushv