espressif/esp-rainmaker #9: WiFi Provisioning for custom phone app

jwktje, 2020-06-17:
How would I go about implementing the WiFi Provisioning in my own phone app?
I would wanna build this phone app with React Native.

The REST API only seems to expose interaction for after the device has been provisioned and mapped to a user. So how would I go about implementing this provisioning/mapping step?

shahpiyushv, 2020-06-17:
@jwktje , there are various ways by which you can go about this

1) Use the ESP RainMaker Phone app sources. Links: [iOS](https://github.com/espressif/esp-rainmaker-ios), [Android](https://github.com/espressif/esp-rainmaker-android).
2) Use the Wi-Fi Provisioning phone libraries in your own app. Links: [iOS](https://github.com/espressif/esp-idf-provisioning-ios), [Android](https://github.com/espressif/esp-idf-provisioning-android).
3) Check the Wi-Fi provisioning protocol [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/wifi_provisioning.html#user-side-implementation) and implement on your own.

jwktje, 2020-06-17:
@shahpiyushv Option 2 looks like a good option. Would it be hard to wrap both the iOS and Android libraries into a React Native Plugin? I guess that's out of scope for this issue so you can close it if you want. But I was just wondering if maybe you have experience with this.

vikas-chandra-mnnit, 2020-06-18:
@jwktje To wrap the libraries some code and data transfer logic needs to be added in the libraries as well. Both our libraries are open source so you can take the source code and add methods in it to exposes the api's to react native module. It seems a bit hard to start with but if you decide to proceed then there is full API documentation for iOS library :https://espressif.github.io/esp-idf-provisioning-ios/. Let us know if you face any problem.

Also if you decide and finish the wrapper then if possible do share your work. It will be helpful for us to give support to React Native developers in future.

jwktje, 2020-06-25:
@vikas-chandra-mnnit @shahpiyushv Thanks for the input. I'm a Webdeveloper by trade so I feel like I'm missing the App-dev skills to make this work. If anyone happens to stumble upon this and knows how to make this wrapper for React Native work, I'm willing to put a code bounty of $100 on this. I know this is probably not much. But for me this is for a hobby project so hopefully any app-devs that read this don't feel insulted.

jwktje, 2020-06-28:
As a side-note update, I did successfully update my prototype to work with provisioning based on [this example](https://trello.com/c/d6c23UEz/45-https-githubcom-espressif-arduino-esp32-tree-master-libraries-wifi-examples-wifiprov). 

It would be great if there was a way to allow end users to provision without needing the ESP SoftAp Prov app. I will try and see if I can find some way to make it work in React Native and will report back if I do.

amoghpalnitkar, 2020-07-01:
@jwktje I'm almost doing the same thing as you are. I will most likely be able to spend some time this week and early next week to get this app migrated to react-native via native modules. let me know if you can wait till then.

jwktje, 2020-07-01:
@amoghpalnitkar That would be amazing! And if there is anything I can do to assist during development or afterwards to test, let me know! I really wanna collaborate on this, but as I mentioned, I know almost nothing of native app development sadly. 

What do you think your approach would be? Would you work on a bridging library that wraps both the [iOS](https://github.com/espressif/esp-idf-provisioning-ios) and the [Android](https://github.com/espressif/esp-idf-provisioning-android) library from Espressif?

amoghpalnitkar, 2020-07-01:
@jwktje Correct. This would be a wrapper around the iOS and Android esp-idf-provisioning library.

A function call from react-native will hand over the control to the native code. The native code will interact with the native esp-idf-provisioning library. This will also have UI to scan the QR code & provision the device. 
On successful provisioning, it will hand back control to react-native.

jwktje, 2020-07-01:
@amoghpalnitkar If you are able to provide such a library for React Native that would be amazing, and that's exactly what I need to move forward with my project. So if you have the time to make that work I will for sure send you that code bounty I mentioned in my previous content.

Let me know if I can do something to help, or if you have it working to a point where I can test it. Thanks in advance for lending your time to the community! :D

amoghpalnitkar, 2020-07-02:
@jwktje It looks like we will need to include the login screens in the library as well, but probably as a react-native plugin rather than native. aws-cognito has decent react-native support so it might make sense to use it.

I am still in two minds whether we should go native or react-native for login pages. @jwktje What do you think? Which use case suits you better?

@shahpiyushv would it be possible for you to share the aws-cognito poolId & clientId?

jwktje, 2020-07-02:
@amoghpalnitkar In that case I think your use-case might differ from mine. I don't need the full scope of the ESP RainMaker platform to be exposed through React Native.

Only the provisioning functions/libraries. And really only SoftAP, no BLE.
So if I could run these functions for Android/iOS through a bridge library I would be happy;

- createESPDevice
- espDevice.connect
- espDevice.scanWifiList
- espDevice.provision

amoghpalnitkar, 2020-07-02:
@jwktje Oh, I see. Does that mean you only need those functions to be exposed and you would be building your own screens in react-native over these functions?

That could still be a subset of my current work.

jwktje, 2020-07-03:
@amoghpalnitkar Yes. Because I want to integrate the WiFi provisioning of my ESP32-based device into the current onboarding/setup steps of my app. So I will build out my own screens for connecting, scanning and so on. 

Sorry that it's a bit confusing, because this issue is open on the esp-rainmaker repository. But I will not de using the RainMaker functions on my device, nor the REST backend they provide. Really only the provisioning side of it. 

Would that still be something you could help out with or does that not really work with how your project is set up?

amoghpalnitkar, 2020-07-03:
@jwktje I can build something that is usable for you, now that I know the exact use case. Will keep you posted.

amoghpalnitkar, 2020-07-05:
@jwktje I have sent you a repo invite to the first draft. Currently, it only has ios bridge. Building the Android bridge can happen once we finalise iOS. Please create issues on that repo for further feedback. Thanks!

jwktje, 2020-07-05:
@amoghpalnitkar I accepted the invite and will start testing tomorrow! Thanks very much! I will create issues on the repo for anything that I find during testing. :D

tsVestor, 2020-07-06:
@amoghpalnitkar I have the exact same use case as @jwktje . I could do the Android testing with React Native.

jwktje, 2020-07-06:
I have started testing and will commit any changes that I think I can make on a separate branch of the repo. This way I can make a PR and @amoghpalnitkar can check them before merging. Maybe @tsVestor can do something similar for any Android quirks? And obviously testing and creating issues is super helpful feedback too!

tsVestor, 2020-07-06:
Sure

tsVestor, 2020-07-19:
@amoghpalnitkar @jwktje could you please invite me to the corresponding repo

jwktje, 2020-07-22:
Hey @tsVestor We are still refactoring it into something that is easier to get more collaborators on board with. Currently it's more a full prototype project and @amoghpalnitkar is working on turning it into a plugin. Once we have that we could share the repo.

tsVestor, 2020-08-25:
Hi @jwktje and @amoghpalnitkar I had a look at your react-native-esp-idf-provisioning repository. It is missing the android implementation, is it under development or did you put it on hold?

tsVestor, 2020-08-27:
@amoghpalnitkar  I'm willing to add a code bounty of 50 USD for the React-Native Android implementation.

amoghpalnitkar, 2020-08-29:
Hey @tsVestor,  apologies that I couldn't reach out earlier. 
Yes, the project is currently missing android implementation. 

I was neck-deep in a few releases over the last couple of weeks so couldn't find time for the Android part as well.

I have switched the repo's visibility to private since I feel it is not yet ready for the public. And I have added you as a collaborator to the repo.

I'll add the android implementation there.

amoghpalnitkar, 2020-09-06:
@tsVestor I have added the android implementation. Could you please help me test the modules out?

tsVestor, 2020-09-06:
@amoghpalnitkar thank you, I tried it immediately. Unfortunately the example crashes in debug and release build immediately after create_device() returns with {"error": "success", "value": undefined}

amoghpalnitkar, 2020-09-06:
Ah, this might just have to do with the android permissions.

Can you please create an issue for this in the project repository?

Also, can you try again after giving the FINE_LOCATION permission manually by going to the device settings?

tsVestor, 2020-09-06:
The FINE_LOCATION permission is granted, I opened an issue

digimatt22, 2020-09-16:
@amoghpalnitkar i would be happy to contribute to the development of this package for react native. I have a couple uses for such a library on upcoming projects and would rather support an open source project that start from scratch. Let me know if you are looking for support in code and testing.

amoghpalnitkar, 2020-09-16:
@digimatt22 Thanks for the response. I would appreciate your help. Sent you an invite.

bbinet, 2020-09-18:
@digimatt22 note that there is also already an existing wifi provisioning app built with Flutter: https://github.com/sunshine-tech/esp_provisioning
Flutter is not react native, but quite similar technology.

Ollebolle, 2020-09-20:
@amoghpalnitkar I would be very interested in your project repo and can try to help out in any way possible, at least by testing it. I would greatly appreciate an invite! :)

png, 2020-09-21:
@amoghpalnitkar I'm sure you're tired of hearing this by now, but I would also love to try out the library and help out if I can. Working on a project that would benefit from it greatly and am trying from scratch, but, of course, it's a little bit of work.

Satyam-code143, 2021-03-18:
@amoghpalnitkar , @jwktje   Hey, just wanted to know how more days might it take to complete `react-native-esp-idf-provisioning` package. I would like to help, but I am not much familiar with any of the package development. I might need some guidance  for contributing.
Thank You!!

manbomb, 2021-06-17:
> @amoghpalnitkar  I'm willing to add a code bounty of 50 USD for the React-Native Android implementation. 

I'm making a module to react native, are you still willing to pay?

manbomb, 2021-06-17:
> @vikas-chandra-mnnit @shahpiyushv Thanks for the input. I'm a Webdeveloper by trade so I feel like I'm missing the App-dev skills to make this work. If anyone happens to stumble upon this and knows [...]

Look at the top post.

manbomb, 2021-06-20:
`npm i react-native-esp-idf-ble-provisioning-rn`

This is the lib that I made. Look that.

rvalentini1, 2021-06-28:
Hi , is there a react-native espress if ble provisioning for ios?
Thanks