espressif/esp-rainmaker #11: Question: BLE services/characteristics support

bbinet, 2020-06-19:
After reading the rainmaker examples, its model of defining **devices** and **attributes** in rainmaker projects seemed very similar to how we can define **services** and **characteristics** in BLE.

So I was wondering if you were planning to develop some interoperability with BLE so that we can also control our esp device locally with bluetooth when internet connection is not available. This feature would be great!

We sometimes need to operate ESP devices without internet connection (even if the manufacturing/commissioning phase may require a temporary internet connection to configure the device).

shahpiyushv, 2020-06-19:
@bbinet , we plan to integrate the [esp_local_control](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_local_ctrl.html) feature of esp-idf into RainMaker so that control would be possible from phones in the same local network, or even over BLE (for ESP32 only) as you said.

bbinet, 2020-06-22:
Thank you @shahpiyushv, that will be very useful.

bbinet, 2020-06-22:
@shahpiyushv any ETA on that feature?

shahpiyushv, 2020-06-27:
@bbinet , since it would require support on the node as well as the phone apps, considering the other tasks planned, I would say that it could take at least a month. If it happens earlier than that, I will drop a message here.

bbinet, 2020-08-26:
@shahpiyushv any news on this feature?

bbinet, 2020-11-30:
@shahpiyushv I've noticed that the [esp_local_control feature of esp-idf has now been implemented into RainMaker](https://github.com/espressif/esp-rainmaker/commit/892e61b32f1051cc79d76061adc3f20f8e0b17de) so that control would be possible from phones in the same local network, this is great, thank you!

Do you know if it is planned to also support BLE transport for esp_local_control in Rainmaker?

shahpiyushv, 2020-11-30:
@bbinet , it is indeed planned, but it may be taken up later as we are are working on some other features right now.

bbinet, 2020-11-30:
Great, thank you.

Fahadk7, 2022-07-16:
Hi @shahpiyushv 
I was experimenting with ESP Rainmaker, and was interested in using this BLE Local control feature

is this feature now live in Rainmaker?

anupam-dubey, 2023-01-23:
I was too wondering if ESP rainmaker, combine BLE local control feature.
is it done?