espressif/esp-rainmaker #12: Receive gpio value from nowhere

CodorStelian, 2020-07-02:
Hi,
I have an issue with gpio state.
Esp32 receive two value from nowhere, it is very strange that it happens before wifi connection and after gpio state appears false in rainmaker app.
[image]

Thanks

shahpiyushv, 2020-07-02:
@CodorStelian have you used  PROP_FLAG_PERSIST for your parameters? If yes, your callback gets invoked as soon as you add the parameter, so that the stored values can reflect on the actual hardware.

CodorStelian, 2020-07-02:
@shahpiyushv Thank you very much.
Just disabled it from esp_maker_core.h and problem is solved.😁

closed 2020-07-02 by CodorStelian

shahpiyushv, 2020-07-02:
@CodorStelian , good to know that the problem got resolved. However, adding this flag is something that the application does, not the core component. So, I did not understand what you mean by "Just disabled it from esp_maker_core.h ". Where exactly did you make the change?

CodorStelian, 2020-07-06:
@shahpiyushv Just changed 1 with 0. I don't know if this is correct solution but it works.
[image]

shahpiyushv, 2020-07-06:
@CodorStelian , this isn't the correct solution. Please check if your code has any esp_rmaker_device_add_param() call with the property flags having PROP_FLAG_PERSIST set. With your change, even the custom device name that you may set from the phone app will change back to the default after a reboot.

CodorStelian, 2020-07-06:
@shahpiyushv , my code doesn't have any PROP_FLAG_PERSIST. If you want to check the code [https://github.com/CodorStelian/ESP32-Rainmaker-22-Z](https://github.com/CodorStelian/ESP32-Rainmaker-22-Z)

shahpiyushv, 2020-07-06:
@CodorStelian , my bad! I should have realised from the original log itself.

For all parameters that have the PROP_FLAG_PERSIST set, the service callback gets invoked once during the initialisation sequence. The "name" parameter, which is added internally, has this flag set by default. Your callback is getting invoked for the Name parameter, as you can see in these lines:

```
app_main: Received value = true for main-light - name
app_main: Received value = false for second-light - name
```

The culprit for this is this code snippet in your `common_callback`

```
else if (app_driver_set_gpio(dev_name, val.val.b) == ESP_OK) {
    ESP_LOGI(TAG, "Received value = %s for %s - %s",
            val.val.b? "true" : "false", dev_name, name);
}
```

This is getting called if the parameter name isn't matching with any of the parameters you are checking. Since the "name" parameter isn't handled in your callback (and it need not be handled anyways), the code is going into this else block.

CodorStelian, 2020-07-06:
I see..., my mistake. Thank you for helping me and have a nice day!