espressif/esp-rainmaker #13: Local Notifications on client phones?

jhubbardsf, 2020-07-06:
Is this something that is being looked into for the future. I haven't noticed anybody mention it and it seems like it would be a very very useful feature. A lot of my devices need to be able to notify the user if a certain condition is met. This would obviously be most useful through the app. Unless I missed something big I don't see this feature mentioned anywhere. And since not all IoT devices are just for the user to control, a lot are for monitoring as well I think it could be useful.

shahpiyushv, 2020-07-06:
@jhubbardsf , there are plans of integrating the [ESP Local Control](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_local_ctrl.html) but not local notifications. This surely would be useful and I will check what can be done to support this. There are 2 options I see:
1) ESP Local Control uses mDNS for service discovery. We can have some state change indicator in the mDNS TXT records. If there is any change, the node will indicate it in mDNS. Clients in the same network can then query the parameters to get new value.
2) We can use the asynchronous communication facility (SSE) of HTTPD and send notifications to all connected clients.

This is not something we may do immediately, but I will add it in the roadmap. And yes, we are always open to contributions if people have any ideas and/or implementations for this :)

jhubbardsf, 2020-07-23:
I'm liking the idea of number 2 more than number 1. I'll be honest, I was a little weary of contributing to mobile app related code with Espressif because the esp-idf-provisioning-ios repository used to have countless bugs that the example app on the store didn't and obviously wasn't the same codebase. Looks like you guys are doing a lot better at keeping them synced. If Rainmaker is gonna be legit supported I'll definitely think about a good way to implement this :)

jhubbardsf, 2020-07-23:
I just want to make sure we're talking about the same think. ESP Local Control allows  a developer to write code that can control an MCU via web, ble, whatever. Which is great for when your internet goes down but you still need to control your devices and wanna seamlessly hop over to BLE. But I'm talking about just plain notification on the app when an event on the MCU is met. I'm not sure it's relevant to ESP Local Control. I'm talking about a way for the Rainmaker to show a popup notification that says "Your Plant Is Too Dry!" and add a badge number to the Rainmaker app on the users phone. Sorry if I misunderstood you.

I think we might be agreeing with each other though and I think option 2 seems like a cleaner approach although there might be an option 3 (to figure out). I was weary about contributing to espressif's mobile codebase's because it seemed like your ios examples were never in sync a year ago which hurt when there were bugs in the github repo but not in the app store example. Now it looks like you guys are doing a great job.

I might take a stab at implementing this and if I find a good way I'll obviously let you know and put in a PR lol. I can see why you mentioned option 1 though. My first instinct was more something like option 2, is there are giant use case for 1? I can see how ESP Local Control is useful (haven't personally used it but cool feature). But doesn't that require the end user to enable mDNS on several levels?

For IIoT, sure. That could be a requirement. But isn't making mDNS enabled on a local level a bit much for more casual IoT uses? That's why I'd learn to 2 or 3rd implementation. Just brain storming.

shahpiyushv, 2020-07-23:
@jhubbardsf , mDNS is actually quite suited for such simpler applications. Even Apple's HomeKit eco system solely relies on mDNS for discovering accessories on the local network. ESP Local Control, in this context, is more as a means of discovery and establishing a secure channel between the client and device. Since it anyways uses BLE or HTTPD, it could very well be modified to send an asynchronous message back on the same secure channel.

jhubbardsf, 2020-07-23:
Interesting, I'll definitely be checking out that codebase for ESP LC this weekend. Since this is what I need to use rainmaker I'll give it a shot and hit you up if I have any specific questions. ESP LC does sound like a good option.