espressif/esp-rainmaker #17: Build Issue

intothefuture20, 2020-08-04:
Not able to build example - Switch and Fan, did not try others. I get the following error in both:

[image]
[image]

Seems something wrong in led_strip under components.

shahpiyushv, 2020-08-04:
@intothefuture20 , I suppose you are using esp-idf v4.0 or v4.1. Can you try with the master branch?

intothefuture20, 2020-08-04:
I tried the master branch and got another error now:

[image]

and similarly for switch.elf

shahpiyushv, 2020-08-04:
@intothefuture20 , the log is insufficient to understand what is going wrong. A few questions:
- Did you try an `idf.py fullclean` before rebuilding after you switched the branches? I would also recommend that you delete the sdkconfig file before rebuilding.
- Are you able to build other esp-idf examples on the master branch?

intothefuture20, 2020-08-04:
Hello Piyush,

May be off the league question, but are there other ways to build the examples? I mean the easier one maybe, easier method?

And yes I was able to build esp-idf examples without any error

intothefuture20, 2020-08-04:
Hello Piyush,

Thank you for your help. Much appreciated. As suggested deleting sdkconfig file helped and the build was successful.

shahpiyushv, 2020-08-04:
@intothefuture20 , good to know that the problem got fixed. If you prefer IDEs, we have a [VS Code extension](https://github.com/espressif/vscode-esp-idf-extension) available for esp-idf. You can also check the [RainMaker Support](https://github.com/espressif/vscode-esp-idf-extension#esp-rainmaker-support) section there.

shahpiyushv, 2020-08-06:
Closing this issue as it has been resolved.

closed 2020-08-06 by shahpiyushv