espressif/esp-rainmaker #18: analogRead function for standard analog input like Arduino?

rtxsc, 2020-08-24:
Hi everyone, may i know if there'll be an update for getting analogRead function like how we would do it in Arduino?

shahpiyushv, 2020-08-24:
@rtxsc , ESP RainMaker in itself does not have any APIs for analog/digital read/write. You will have to use underlying esp-idf APIs. For analog read, please refer [this document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html). Meanwhile, [RainMaker + Arduino](https://github.com/espressif/arduino-esp32/pull/4194) is also in works

rtxsc, 2020-08-24:
> @rtxsc , ESP RainMaker in itself does not have any APIs for analog/digital read/write. You will have to use underlying esp-idf APIs. For analog read, please refer [this [...]

Thanks for the response. Can’t wait for the implementation in Arduino environment. Keep up the good work sir. ESP Rainmaker as of now is a great platform for prototyping IoT device such as TP-link KASA wifi smart-plug. I could see the potential there. The only let down now is the scarcity of library support for Arduino-like functionality and expansion.