espressif/esp-rainmaker #21: ESP32

boujmilmaha, 2020-09-09:
How can i get to connect through a ESP32 with RainMaker . i didn't get it clearly using the documentations , can i get some totorials for the steps that i could use to setup the CLI and how i get the QR code when i flash the example of a switch . 
i get this two error in the monitor every time i flash the example
----E (870) esp_rmaker_core: Failed to initialise Node Id. Please perform "claiming" using RainMaker CLI.
----E (870) app_main: Could not initialise node. Aborting!!!
how can i solve them !!!!!!!!

shahpiyushv, 2020-09-09:
@boujmilmaha have you set up the CLI using the steps given [here](https://rainmaker.espressif.com/docs/cli-setup.html) and tried host driven claiming as per this [doc](https://rainmaker.espressif.com/docs/claiming.html#host-driven-claiming-esp32)?

boujmilmaha, 2020-09-09:
yes i setup the CLI but i can't tried host driven claiming i found this error .
usr/bin/env : "python3".  permission dinied (when i wrote :     ./rainmaker.py login        )

shahpiyushv, 2020-09-09:
@boujmilmaha Can you try this instead?

```
cd /path/to/esp-rainmaker/cli
$ python3 rainmaker.py login
```

boujmilmaha, 2020-09-09:
i try this :  $ PY rainmaker.py login
this is the error i got :
Traceback (most recent call last):
  File "rainmaker.py", line 19, in <module>
    from rmaker_cmd.node import get_nodes, get_node_config, get_node_status,\
  File "C:\esp-rainmaker\cli\rmaker_cmd\node.py", line 20, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'

________________________________
De : boujmil maha <boujmilmaha534@outlook.fr>
Envoyé : mercredi 9 septembre 2020 12:16
À : espressif/esp-rainmaker <reply@reply.github.com>
Objet : RE: [espressif/esp-rainmaker] ESP32 (MEGH-1256) (#21)

shahpiyushv, 2020-09-09:
Seems you may have missed the ["Installing Dependencies"](https://rainmaker.espressif.com/docs/cli-setup.html#installing-dependencies) step, specifically this:

```
$ cd /path/to/esp-rainmaker/cli/
$ pip3 install -r requirements.txt
```

boujmilmaha, 2020-09-09:
i do it   :   $ pip3 install -r requirements.txt

then ,i got this :
Defaulting to user installation because normal site-packages is not writeable
Processing c:\users\infor\desktop\esp-idf\tools\kconfig_new\esp-windows-curses
Requirement already satisfied: setuptools>=21 in c:\program files\python38\lib\site-packages (from -r requirements.txt (line 4)) (47.1.0)
Requirement already satisfied: click>=5.0 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 8)) (7.1.2)
Requirement already satisfied: pyserial>=3.0 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 11)) (3.1)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 13)) (0.26)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in c:\users\infor\appdata\roaming\python\python38\site-packages (from cryptography>=2.1.4->-r requirements.txt (line 11)) (1.14.2)
Requirement already satisfied: six>=1.4.1 in c:\users\infor\appdata\roaming\python\python38\site-packages (from cryptography>=2.1.4->-r requirements.txt (line 11)) (1.15.0)
Requirement already satisfied: pycparser in c:\users\infor\appdata\roaming\python\python38\site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r requirements.txt (line 11)) (2.20)
Building wheels for collected packages: esp-windows-curses
  Building wheel for esp-windows-curses (setup.py): started
  Building wheel for esp-windows-curses (setup.py): finished with status 'done'
  Created wheel for esp-windows-curses: filename=esp_windows_curses-0.1-py3-none-any.whl size=1146 sha256=d6d8527a607f19b00e6252f169c1845311d817a6442d8b233d4e25842663f2c8
  Stored in directory: c:\users\infor\appdata\local\pip\cache\wheels\8f\f5\99\d623842dd9259d5e32431613481b744e3a7f005b320a98cd3e
Successfully built esp-windows-curses
Installing collected packages: esp-windows-curses
  Attempting uninstall: esp-windows-curses
    Found existing installation: esp-windows-curses 0.1
    Uninstalling esp-windows-curses-0.1:
      Successfully uninstalled esp-windows-curses-0.1
Successfully installed esp-windows-curses-0.1

boujmilmaha, 2020-09-09:
this is the error i got when i install  the Dependencies<https://rainmaker.espressif.com/docs/cli-setup.html#installing-dependencies>
 ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly

________________________________
De : boujmil maha <boujmilmaha534@outlook.fr>
Envoyé : mercredi 9 septembre 2020 13:12
À : espressif/esp-rainmaker <reply@reply.github.com>
Objet : RE: [espressif/esp-rainmaker] ESP32 (MEGH-1256) (#21)

i do it   :   $ pip3 install -r requirements.txt

then ,i got this :
Defaulting to user installation because normal site-packages is not writeable
Processing c:\users\infor\desktop\esp-idf\tools\kconfig_new\esp-windows-curses
Requirement already satisfied: setuptools>=21 in c:\program files\python38\lib\site-packages (from -r requirements.txt (line 4)) (47.1.0)
Requirement already satisfied: click>=5.0 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 8)) (7.1.2)
Requirement already satisfied: pyserial>=3.0 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 11)) (3.1)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 13)) (0.26)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in c:\users\infor\appdata\roaming\python\python38\site-packages (from cryptography>=2.1.4->-r requirements.txt (line 11)) (1.14.2)
Requirement already satisfied: six>=1.4.1 in c:\users\infor\appdata\roaming\python\python38\site-packages (from cryptography>=2.1.4->-r requirements.txt (line 11)) (1.15.0)
Requirement already satisfied: pycparser in c:\users\infor\appdata\roaming\python\python38\site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r requirements.txt (line 11)) (2.20)
Building wheels for collected packages: esp-windows-curses
  Building wheel for esp-windows-curses (setup.py): started
  Building wheel for esp-windows-curses (setup.py): finished with status 'done'
  Created wheel for esp-windows-curses: filename=esp_windows_curses-0.1-py3-none-any.whl size=1146 sha256=d6d8527a607f19b00e6252f169c1845311d817a6442d8b233d4e25842663f2c8
  Stored in directory: c:\users\infor\appdata\local\pip\cache\wheels\8f\f5\99\d623842dd9259d5e32431613481b744e3a7f005b320a98cd3e
Successfully built esp-windows-curses
Installing collected packages: esp-windows-curses
  Attempting uninstall: esp-windows-curses
    Found existing installation: esp-windows-curses 0.1
    Uninstalling esp-windows-curses-0.1:
      Successfully uninstalled esp-windows-curses-0.1
Successfully installed esp-windows-curses-0.1

boujmilmaha, 2020-09-09:
$ python rainmaker.py claim COM7
I HAVE THIS PROBLEM

[ERROR]:Failed to connect to Espressif device: Timed out waiting for packet header
Failed to connect to Espressif device: Timed out waiting for packet header

________________________________
De : boujmil maha <boujmilmaha534@outlook.fr>
Envoyé : mercredi 9 septembre 2020 13:24
À : espressif/esp-rainmaker <reply@reply.github.com>
Objet : RE: [espressif/esp-rainmaker] ESP32 (MEGH-1256) (#21)

this is the error i got when i install  the Dependencies<https://rainmaker.espressif.com/docs/cli-setup.html#installing-dependencies>
 ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly

________________________________
De : boujmil maha <boujmilmaha534@outlook.fr>
Envoyé : mercredi 9 septembre 2020 13:12
À : espressif/esp-rainmaker <reply@reply.github.com>
Objet : RE: [espressif/esp-rainmaker] ESP32 (MEGH-1256) (#21)

i do it   :   $ pip3 install -r requirements.txt

then ,i got this :
Defaulting to user installation because normal site-packages is not writeable
Processing c:\users\infor\desktop\esp-idf\tools\kconfig_new\esp-windows-curses
Requirement already satisfied: setuptools>=21 in c:\program files\python38\lib\site-packages (from -r requirements.txt (line 4)) (47.1.0)
Requirement already satisfied: click>=5.0 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 8)) (7.1.2)
Requirement already satisfied: pyserial>=3.0 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 11)) (3.1)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in c:\users\infor\appdata\roaming\python\python38\site-packages (from -r requirements.txt (line 13)) (0.26)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in c:\users\infor\appdata\roaming\python\python38\site-packages (from cryptography>=2.1.4->-r requirements.txt (line 11)) (1.14.2)
Requirement already satisfied: six>=1.4.1 in c:\users\infor\appdata\roaming\python\python38\site-packages (from cryptography>=2.1.4->-r requirements.txt (line 11)) (1.15.0)
Requirement already satisfied: pycparser in c:\users\infor\appdata\roaming\python\python38\site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r requirements.txt (line 11)) (2.20)
Building wheels for collected packages: esp-windows-curses
  Building wheel for esp-windows-curses (setup.py): started
  Building wheel for esp-windows-curses (setup.py): finished with status 'done'
  Created wheel for esp-windows-curses: filename=esp_windows_curses-0.1-py3-none-any.whl size=1146 sha256=d6d8527a607f19b00e6252f169c1845311d817a6442d8b233d4e25842663f2c8
  Stored in directory: c:\users\infor\appdata\local\pip\cache\wheels\8f\f5\99\d623842dd9259d5e32431613481b744e3a7f005b320a98cd3e
Successfully built esp-windows-curses
Installing collected packages: esp-windows-curses
  Attempting uninstall: esp-windows-curses
    Found existing installation: esp-windows-curses 0.1
    Uninstalling esp-windows-curses-0.1:
      Successfully uninstalled esp-windows-curses-0.1
Successfully installed esp-windows-curses-0.1

shahpiyushv, 2020-09-09:
@boujmilmaha , is the same COM port open somewhere else as well? Probably in `idf.py monitor`?

shahpiyushv, 2020-10-12:
Closing this issue as there has been no activity on it. Please feel free to reopen if required.

closed 2020-10-12 by shahpiyushv