espressif/esp-rainmaker #22: claim <port>

boujmilmaha, 2020-09-09:
i got an error every time i do this step : 
$ python rainmaker.py claim 'COM7'

[ERROR]:Failed to connect to Espressif device: Timed out waiting for packet header
Failed to connect to Espressif device: Timed out waiting for packet header

what should i do !!

shahpiyushv, 2020-09-10:
@boujmilmaha , did you read my comment [here](https://github.com/espressif/esp-rainmaker/issues/21#issuecomment-689556390)? Is your COM port open somewhere else, probably in idf.py monitor? Is the COM port correct? Have you tried removing the quotes or adding double quotes to COM7?

shahpiyushv, 2020-10-12:
Closing this issue as there has been no response on this, and the "Assisted Claiming" would also help mitigate it.

closed 2020-10-12 by shahpiyushv