espressif/esp-rainmaker #23: GPIO isr service already installed

SNSIOTP, 2020-09-16:
Dear All,

I am getting the following warning/error when I flash and monitor on ESP32.

[image]

I have used 8 GPIO pins of ESP32. 4 for output and 4 for input.

Seeking assistance please

shahpiyushv, 2020-09-16:
@SNSIOTP , can you also share your code snippet indicating the GPIO numbers?

SNSIOTP, 2020-09-17:
[image]

sachin0x18, 2020-09-17:
@SNSIOTP IO6-IO11 are connected to the on board SPI flash so they aren't available for general purpose usage. Please use some other GPIOs.

SNSIOTP, 2020-09-17:
[image]

ameyinamdar, 2020-09-17:
Hi @SNSIOTP,
Please check which module you are using and refer to corresponding pin definition in the module datasheet. For example, if WROOM-32 module is being used, please refer to https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32_datasheet_en.pdf

sachin0x18, 2020-09-17:
Refer to section [3.2 External Flash and SRAM](https://www.espressif.com/sites/default/files/documentation/esp32-wroom-32d_esp32-wroom-32u_datasheet_en.pdf)

SNSIOTP, 2020-09-17:
It seems I have to procure few genuine modules manufactured by Espressif to be able to test efficiently and will be more reliable. With these knock offs GPIO structure seems to be unreliable.

Thank you for all the suggestions and help. Issue can ne closed. Will test again and if the issue still persist will submit an issue.

SNSIOTP, 2020-09-17:
Or maybe reopen the same issue.

closed 2020-09-17 by SNSIOTP