espressif/esp-rainmaker #24: Self Claiming for ESP32

bbinet, 2020-09-22:
After reading [claiming docs](https://rainmaker.espressif.com/docs/claiming.html), I understand that  ESP32-S2 use a secret key programmed in efuse during chip manufacturing to allow self-claiming, but I don't really understand why this secret key is important in the self claiming process.
Actually, Assisted Claiming seems very similar to Self Claiming but does not seem to use any secret key.
So can you explain what prevents ESP32 from having support for Self Claiming and how the ESP32-S2 secret key is used in the cloud claiming service ?

shahpiyushv, 2020-09-22:
@bbinet The primary difference is that ESP32-S2 directly authenticates itself with the claiming service (using the challenge-response created with the secret key), whereas in case of ESP32, it is the user that gets authenticated. So, you can pair as many ESP32-S2s as you want using self-claiming, but only 5 ESP32s with assisted claiming, as that is the quota assigned to each user. Since ESP32-S2 is a newer chip, we could ensure that the secret key is programmed in all chips, but for ESP32, which is already in production since long, we cannot do so.

bbinet, 2020-09-22:
@shahpiyushv Would it be possible to use base MAC address which is factory-programmed (that is returned by esp_efuse_mac_get_default functoin) on ESP32 as a replacement to the secret key programmed in the ESP32-S2 to use self claiming on ESP32?

ameyinamdar, 2020-09-22:
Hi @bbinet MAC address is not secret and serially allocated. So it's quite easy to claim on behalf of some other device if MAC address as used as a replacement of secret key.

May I know what you are trying to achieve here? There are also other claiming methods possible like pre-provisioned modules.

bbinet, 2020-09-22:
@ameyinamdar Sure, I try to better understand the claiming process as I would eventually like to implement our own claiming service for esp32 similarly to rainmaker cloud implementation so that I can keep control over our esp32 devices using our own servers.

Actually it would be great to have opensource implementations of the rainmaker cloud enpoints.

I would be interested to know more about the other claiming methods possible like pre-provisioned modules you are talking about.

ameyinamdar, 2020-09-23:
@bbinet AWS recently announced general availability of a feature "multiaccount registration" - https://aws.amazon.com/about-aws/whats-new/2020/04/simplify-iot-device-registration-and-easily-move-devices-between-aws-accounts-with-aws-iot-core-multi-account-registration/ 

With this feature, any device whose certificate is registered in the AWS account can connect to AWS-IoT using the same certificate. So you can use any way to generate certificates (and private keys) and program them on the device. With rainmaker, we will provide scripts to generate batch of certificates. Espressif also provides pre-provisioned modules that come with pre-programmed certificates and these certificates software manifest is provided to customers who can take advantage of multiaccount registration feature.

Claiming is primarily development use-case whereby the methods mentioned above are useful when you want to connect larger number of devices to Rainmaker cloud.

bbinet, 2020-09-23:
Thank you @ameyinamdar for these informations, this is interesting.

closed 2020-09-23 by bbinet