espressif/esp-rainmaker #25: Google Play: "Your device isn't compatible with this version."

seedsofhope, 2020-10-09:
Come on, guys... What makes a device incompatile?

khushbushah2302, 2020-10-09:
@seedsofhope ,

Can you please give some detail ? 
1. What is the android version of your phone ? 
2. Is app already installed on phone and currently giving this error Play Store ?

Also check [this Google Play Store issue](https://support.google.com/googleplay/thread/5241214?hl=en), if it is a problem of Google Play store.

copercini, 2020-10-15:
I'm with this same issue, 
My android is 6.0.1 and I'm not able to download it from play store due "Your device isn't compatible with this version."

khushbushah2302, 2020-10-16:
Unfortunately, we only support Android 7 and later. It is mentioned on [esp-rainmaker-android](https://github.com/espressif/esp-rainmaker-android#supports) repo, but we will see that it is mentioned elsewhere was well

khushbushah2302, 2020-10-16:
Hi @copercini ,

We are trying to give support for Android 6 version. Here is the [apk](https://drive.google.com/file/d/16W2OdwHrO_5NYwCgWRxYhR_BqEhLNymH/view?usp=sharing). Try this app and let us know how it goes.

copercini, 2020-10-16:
@khushbushah2302 Thanks, it's working fine on my android 6.0.1 =)
I tested with an ESP32-S2 dev kit and Switch example firmware, everything seems working as expected

shahpiyushv, 2021-01-18:
Closing this issue as it has been addressed.

closed 2021-01-18 by shahpiyushv