espressif/esp-rainmaker #28: Alexa Temperature

NemesisXB, 2020-10-23:
Rainmaker temperature devices do not show up in Alexa. I have other devices that show temperature in Alexa, so it should be able to support it.

shahpiyushv, 2020-10-23:
@NemesisXB , currently, only switch and lightbulb are supported. We will go on adding more categories, prioritising the ones that people are looking for. Since temperature sensor is mentioned as "Standard type", I guess it was natural to assume that it would be supported, but unfortunately, it is not the case. I will add a note in the docs highlighting this and also consider adding temperature sensor to our alexa skill asap.

iz0eyj, 2020-12-10:
Forgive me if I intrude on your discussion.
Unlike commercial IoT components, this very important Espressif initiative gives us the opportunity to create objects designed specifically for our automation needs (often only approximable with what is available on the market).
However, the approach should be different, let's say opposite, to that normally followed in the IoT, and this EXACTLY because the possibility of custom programming the components would make the number of possible objects so large that it cannot be managed in detail.
The only possibility I see is that Espressif gives us highly customizable standard interfaces (both in the icon and in the number of channels), then leaving the developer responsible for creating the code.
 In particular, it would be necessary to provide generic interfaces for:
 - Digital input devices
 - Digital output devices
 - Analog input devices
 - Analog output devices
 - Flag devices *
 - Counter devices *
 - Timer devices *

(* On this I will open a special thread.)

Once the standard interface has been established and the possibility of customizing the icon, it will no longer matter if an analog device will be a thermometer rather than a fluid level meter, and the same will apply to a digital output, which can indifferently be a  light bulb or switch that controls a socket.

 My three cents, Federico.

NemesisXB, 2020-12-10:
> Forgive me if I intrude on your discussion.
> Unlike commercial IoT components, this very important Espressif initiative gives us the opportunity to create objects designed specifically for our [...]

I think you picked the wrong issue to piggyback on. This is not a limitation by espressif/rainmaker. Alexa only supports a handful of types. This issue was raised because both alexa and rainmaker supports temperature types, but even though creating a temperature type in rainmaker, it did not show up in alexa. As far as I understand, none of the types you mentioned is supported by Alexa.

shahpiyushv, 2020-12-10:
@iz0eyj , as @NemesisXB mentioned, this is a somewhat different requirement and would need a separate thread. However, we indeed have added support for the generic Toggle and Range Controllers that Alexa supports and it should be available soon.

iz0eyj, 2020-12-10:
@shahpiyushv I opened the thread.
@NemesisXB in home automation, what is implemented at the component level and Alexa management are different levels, with different degrees of abstraction from the real hardware and different support needs.
 Alexa (or G Assistant) can do few things through their Routines, but since they are ultimately simple macros, in practice they are very useful voice interfaces.
 Without a voice interface, home automation is of little use, but it is not possible to implement complex algorithms that manage complex home automation in Alexa or G Assistant.

shahpiyushv, 2021-03-18:
@NemesisXB Temperature sensor is now supported in the Alexa Skill. Please check. Closing this issue now.

closed 2021-03-18 by shahpiyushv