espressif/esp-rainmaker #29: Select icon to display in mobile app

NemesisXB, 2020-10-23:
It would be a great enhancement if it is possible to specify what icon to use in the mobile app. Basically a list of default images we can select to display when we create the device in rainmaker firmware. For example, I want to switch a pump, so I use a switch device, but would like the icon to be of a pump or similar.

CodorStelian, 2021-10-05:
You can achieve this by creating a new custom device and set the switch param as the main param. Also, you need to create a new icon and build your own phone app, just add your own icon [here](https://github.com/espressif/esp-rainmaker-android/blob/8c58b4e9c886617cbfe556bee95a299e584ba96c/app/src/main/java/com/espressif/ui/adapters/EspDeviceAdapter.java#L70) (for android). Anyway, I think more icons and devices are coming in the near future.

NemesisXB, 2021-10-05:
> You can achieve this by creating a new custom device and set the switch param as the main param. Also, you need to create a new icon and build your own phone app, just add your own icon [...]

Will this not replace the switch icon for any device using it?

CodorStelian, 2021-10-05:
No. I will give you an example for humid sensor which I have implemented in IO Connect
In EspDeviceAdapter.java:
` else if (AppConstants.ESP_DEVICE_HUMID_SENSOR.equals(device.getDeviceType())) {`
`deviceVh.ivDevice.setImageResource(R.drawable.rmk_ic_device_humid_sensor);`
 `}`
and in AppConstants.java
`    public static final String ESP_DEVICE_HUMID_SENSOR = "esp.device.humidity-sensor";
`

NemesisXB, 2021-10-06:
> No. I will give you an example for humid sensor which I have implemented in IO Connect In EspDeviceAdapter.java: ` else if (AppConstants.ESP_DEVICE_HUMID_SENSOR.equals(device.getDeviceType())) {` [...]

Thanks. I'll try this.