espressif/esp-rainmaker #31: Time Zone setting in the device

sivansanthoshd, 2020-11-12:
I am trying to set the time zone for my device as IST by editing the code as per the documentation - https://rainmaker.espressif.com/docs/time-service.html#time-service. 
First i tried to add the following timezone region:
>const char *tz = "Asia/Kolkata";
>esp_err_t esp_rmaker_time_set_timezone(tz);
and the second one using the posix timezone:
>const char *tz_posix = "IST-5:30";
>esp_err_t esp_rmaker_time_set_timezone_posix(tz_posix);
both throw the error 
"../main/app_main.c: In function 'app_main':
../main/app_main.c:161:44: error: expected declaration specifiers or '...' before '*' token
     esp_err_t esp_rmaker_time_set_timezone(*tz);
                                            ^
[1212/1219] Linking CXX static library esp-idf\esp_rainmaker\libesp_rainmaker.a
ninja: build stopped: subcommand failed.
ninja failed with exit code 1"
What is the issue here? If not Am i doing this in a wrong manner?

shahpiyushv, 2020-11-12:
@sivansanthoshd the primary culprit here is the `esp_err_t` preceding the function call. `esp_err_t` is just the return type for the functions. You have to simply call the function like this:

```c
esp_rmaker_time_set_timezone("Asia/Kolkata");
```

sivansanthoshd, 2020-11-12:
After changing it to this

>esp_rmaker_time_set_timezone("Asia/Kolkata");

>I get this error

>../main/app_main.c: In function 'app_main':
../main/app_main.c:159:5: error: implicit declaration of function 'esp_rmaker_time_set_timezone'; did you mean 'esp_rmaker_node_get_info'? [-Werror=implicit-function-declaration]
     esp_rmaker_time_set_timezone(&timezone);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
     esp_rmaker_node_get_info
cc1.exe: some warnings being treated as errors
[1212/1219] Linking CXX static library esp-idf\esp_rainmaker\libesp_rainmaker.a
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

shahpiyushv, 2020-11-12:
@sivansanthoshd to use the `esp_rmaker_time_set_timezone ()` function, you have to include the  `esp_rmaker_utils.h` file in your code.

sivansanthoshd, 2020-11-12:
@shahpiyushv that worked like a breeze. Thank you.

shahpiyushv, 2020-11-17:
Closing this issue since it has been addressed.

closed 2020-11-17 by shahpiyushv