espressif/esp-rainmaker #32: Configuring Multiple devices GPIO in a singlee ESP32

sivansanthoshd, 2020-11-12:
Where to set the GPIO for multiple_device example. Only OUTPUT 19 is available to be configured in the app_driver.c file.
And is there any way to change the icons that is displayed in the mobile application for the specific device type. For example, switch is a common one where i need to display a pump for example. Is it possible? or it will be updated in the future?

shahpiyushv, 2020-11-12:
@sivansanthoshd , the multi device example primarily just demonstrates the software aspect of RainMaker. Hardware and driver integration is something that users will have to do themselves as it will change as per the use cases.

For custom icons, you can raise a feature request and we can take that up when we find appropriate. Else, you have an option to take the [phone app sources](https://rainmaker.espressif.com/docs/quick-links.html#phone-apps) and add your own custom icons.

shahpiyushv, 2021-11-24:
As mentioned, please file feature requests if you want specific icons. Closing this issue since there's no specific following up left as such.

closed 2021-11-24 by shahpiyushv