espressif/esp-rainmaker #34: Error While building

alexiskyriakou, 2020-11-20:
Hello guys,

I have install ESP-IDF and i can build its examples projects then i download the ESP Rainmaker and extract in on me C:/.

To build switch ESP Rainmaker Example, i am opening the ESP-IDF Command prompt, i change directory to switch directory and executing idf.py build and its start building but it gives me the following error

> Using Python in C:\Python38\
> Python 3.8.6
> Using Git in C:\Program Files\Git\cmd\
> git version 2.28.0.windows.1
> Setting IDF_PATH: C:\esp-idf
> 
> Adding ESP-IDF tools to PATH...
>     C:\Users\Alexis\.espressif\tools\xtensa-esp32s2-elf\esp-2020r2-8.2.0\xtensa-esp32s2-elf\bin
>     C:\Users\Alexis\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
>     C:\Users\Alexis\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
>     C:\Users\Alexis\.espressif\tools\idf-exe\1.0.1\
>     C:\Users\Alexis\.espressif\tools\ccache\3.7\
>     C:\Users\Alexis\.espressif\python_env\idf4.1_py3.8_env\Scripts
> 
> Checking if Python packages are up to date...
> Python requirements from C:\esp-idf\requirements.txt are satisfied.
> 
> Done! You can now compile ESP-IDF projects.
> Go to the project directory and run:
> 
>   idf.py build
> 
> 
> C:\esp-idf>cd C:\esp-rainmaker\examples\switch
> 
> C:\esp-rainmaker\examples\switch>idf.py build
> Executing action: all (aliases: build)
> Running cmake in directory c:\esp-rainmaker\examples\switch\build
> Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=1 c:\esp-rainmaker\examples\switch"...
> Warn about uninitialized values.
> -- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.28.0.windows.1")
> -- IDF_TARGET not set, using default target: esp32
> -- ccache will be used for faster recompilation
> -- The C compiler identification is GNU 8.2.0
> -- The CXX compiler identification is GNU 8.2.0
> -- The ASM compiler identification is GNU
> -- Found assembler: C:/Users/Alexis/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
> -- Check for working C compiler: C:/Users/Alexis/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
> -- Check for working C compiler: C:/Users/Alexis/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: C:/Users/Alexis/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
> -- Check for working CXX compiler: C:/Users/Alexis/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Project version: 1.0
> -- Building ESP-IDF components for target esp32
> Loading defaults file C:/esp-rainmaker/examples/switch/sdkconfig.defaults...
> C:\Users\Alexis\AppData\Local\Temp\confgen_tmp8z3uhi6e:2 line was updated to CONFIG_PARTITION_TABLE_SINGLE_APP=n
> C:\Users\Alexis\AppData\Local\Temp\confgen_tmp8z3uhi6e:3 line was updated to CONFIG_PARTITION_TABLE_TWO_OTA=n
> -- Found PythonInterp: C:/Users/Alexis/.espressif/python_env/idf4.1_py3.8_env/Scripts/python.exe (found version "3.8.6")
> -- Could NOT find Perl (missing: PERL_EXECUTABLE)
> -- Adding linker script C:/esp-rainmaker/examples/switch/build/esp-idf/esp32/esp32_out.ld
> -- Adding linker script C:/esp-idf/components/esp32/ld/esp32.project.ld.in
> -- Adding linker script C:/esp-idf/components/esp32/ld/esp32.peripherals.ld
> -- Adding linker script C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
> -- Adding linker script C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
> -- Adding linker script C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
> -- Adding linker script C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
> -- Adding linker script C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
> -- Adding linker script C:/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
> CMake Error at C:/esp-idf/tools/cmake/component.cmake:302 (message):
>   Include directory
>   'C:/esp-rainmaker/components/json_parser/upstream/include' is not a
>   directory.
> Call Stack (most recent call first):
>   C:/esp-idf/tools/cmake/component.cmake:470 (__component_add_include_dirs)
>   C:/esp-rainmaker/components/json_parser/CMakeLists.txt:1 (idf_component_register)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "C:/esp-rainmaker/examples/switch/build/CMakeFiles/CMakeOutput.log".
> cmake failed with exit code 1
> 
> C:\esp-rainmaker\examples\switch>

Can anyone help with this?

Thanks,
Alexis

shahpiyushv, 2020-11-20:
@alexiskyriakou it seems you did not clone esp-rainmaker using the --recursive option. Please check [here](https://github.com/espressif/esp-rainmaker#get-esp-rainmaker). Since you have already cloned the repo, you may use `git submodule update --init --recursive` to recursively update the submodules which will pull in the missing files.

alexiskyriakou, 2020-11-22:
@shahpiyushv thank you issue solved 👍

shahpiyushv, 2020-11-24:
@alexiskyriakou , good to know. Closing this issue now.

closed 2020-11-24 by shahpiyushv