espressif/esp-rainmaker #36: local control with multiple wifi access point network !!

kebiromar, 2020-12-07:
hello I'm working with esp rainmaker and i tested the new feature " local control " and i find something weird.
when the phone and the esp device are connected to  the same WiFi access point the app can discover the esp device.
but when i connect the phone app to different access point but in the same local network the phone app can't reach anymore the esp device.

i don't now if can call this an issue but it still something wrong because a local network is not always with single WiFi access it can be one or more

shahpiyushv, 2020-12-07:
@kebiromar , it actually depends on your Wi-Fi access point settings. The local control feature uses mDNS for device discovery. If your access points are part of a common subnet and allow multicast UDP packets to flow across networks, then the devices should be discoverable. Can you double check?

shahpiyushv, 2021-01-12:
@kebiromar , closing this issue as there hasn't been any feedback. If you are still facing the issue, please feel free to reopen.

closed 2021-01-12 by shahpiyushv