espressif/esp-rainmaker #37: Alexa skill in Italy

iz0eyj, 2020-12-09:
Hi guys,
first of all congratulations for the fantastic job you are doing.
Please note that the Alexa skill is not available in Italy, which makes it a bit problematic to use this system outside of simple tests.
Would it be possible to solve the problem?

 Thanks, Federico

shahpiyushv, 2020-12-09:
@iz0eyj, thanks for reporting this. We will check and get this resolved as soon as possible.

iz0eyj, 2020-12-09:
Thanks @shahpiyushv , if you need help translating the skill into Italian (I don't think it's mandatory), just ask.

iz0eyj, 2021-01-04:
@shahpiyushv hi, still disabled for our country.

[image]

iz0eyj, 2021-01-04:
[image]

iz0eyj, 2021-01-04:
@shahpiyushv Hello, still unaivalable in our country.

[image]

shahpiyushv, 2021-01-15:
@iz0eyj , we are almost done with support for additional languages as well as support for some additional device categories. Once the skill re-certification is done, I will give an update here. Thanks for your patience!

iz0eyj, 2021-01-17:
Thanks @shahpiyushv

shahpiyushv, 2021-01-28:
@iz0eyj, the skill should now be available in your region and language. Can you check?

shahpiyushv, 2021-03-04:
Closing this issue since this has been resolved. Do let us know if you are still facing the problem.

closed 2021-03-04 by shahpiyushv