espressif/esp-rainmaker #38: Suggerimento, dispositivi virtuali

iz0eyj, 2020-12-10:
I am fortunate to have an extremely extensive home automation system, probably one of the most complete in my country (Italy), and for this reason I realize the extreme importance of RainMaker.
 Implementing a pervasive home automation using standard components is not possible;  you can reach a good approximation of what you want but without the possibility of writing custom code INSIDE the single components the approximations will always be ... approximations.
 RainMaker is the answer, that's exactly what was needed!
 Starting from this consideration, I would like to suggest the support of three categories of entirely software devices, i.e. not dedicated to the management / reading of the status of physical machines but to the provision of services to the various components that manage home automation.
 I'll give you a trivial example: the support for algorithms that have to choose different paths based on whether a certain event has or has not occurred IN THE PAST (let's say it has rained or not) in the decision to activate or not activate the  watering some flower pots.
 Well, using standard components (Sonoff, Meross ... whatever you want) is a disaster because there are no components designed for this purpose.  The only solution is to buy relay cards, which have at most four channels, and use them as flags.
 And if the information to be memorized is not digital but analogue ... well it's over because managing it becomes very complicated.
 Starting from this reality, with which I have encountered many times, I suggest to introduce the support of three categories of "virtual" devices: digital flags (bool), analog flags (integer) and timers.
 These devices could be understood as global variables that can be managed in reading and writing by all the components that make up the home automation system, just as the global variables in an application are visible to all the routines that implement them.
 In fact, a trigger category would also be needed, but once the support for what I suggested is implemented it could be created using what is already present.

In summary, my idea is to have support for virtual devices that implement ... let's say 256 (or 1024, they are virtual and cost nothing) variables that can be read by all the other ESP32s on the network.
 In the initial phase, free and common access would be fine, later on we can think about the introduction of protection mechanisms and scope.

 Thanks for your attention, Federico

shahpiyushv, 2020-12-22:
@iz0eyj , sorry for the late reply. If my understanding is correct, you are looking for a network of smart home devices (ESP32s in this case) which operate independently of other clients like phone apps or hubs, right? Simplest example I can think of is an ESP32 based motion sensor turning on a light directly. Going further, these could be virtual devices (as you are proposing), wherein say an ESP32 is not a motion sensor per-se, but a boolean flag and there could be multiple lights on the network waiting for the flag to change.

Is this right?

iz0eyj, 2021-01-04:
Exactly.  When dealing with an extended IoT, one needs to write asynchronous code, i.e. code in which the occurrence of an event differs temporally from its use.  In addition, it is necessary to maintain a state memory relating to the whole of the controlled system.  For example, my home automation behaves differently when I am / I am not at home, and the sequence of events implemented will also be different, with the same input, if the condition in which I went to bed and I fell asleep is verified.  .  In practice, "glib registers" are needed that can be driven and read by any part of the code that implements the IoT.  However components totally absent from the market.  In the case of digital flags, it makes up for with objects similar to the one in the picture, with some of those plates capable of storing 4 bits (born to drive 4 relays), which is to say that to represent the 29 flags used by my home automation consumption 8  IP addresses and I complicate my life by having to take into account WHICH plate contains the bit I need.  And if I need to memorize an analog state ... well I can't help it because there is simply no suitable component.  Having the support of good-sized digital and analogue registers uniquely accessible to a certain IP address would greatly simplify the implementation of complex IoT systems.

Thanks

[image]