espressif/esp-rainmaker #39: [Check node error] API Defintions for Rainmaker Backend Service

victorK8, 2020-12-17:
Hi, 

I'm developing my own web service based on ESP Rainmaker API. I'm with Swagger API, Version 1 (v1) 
The next link shows info. about API: https://swaggerapis.rainmaker.espressif.com/#/

I can login and keep active the login without problems. Next lines show url and succesful logs.

**URL**: 
 _https://api.rainmaker.espressif.com/v1/login_

**LOGS**: 
        _{
          "status": "success",
          "description": "Login successful",
          "idtoken": "idtoken",
          "accesstoken": "accesstoken",
          "refreshtoken": "refreshtoken"
       }_

However, when I try to access the params from a node, it returns errors. Next lines show url and logs.

**URL**: 
_https://api.rainmaker.espressif.com/v1/user/nodes/params?nodeid=[Node Id Value]_

**LOGS**: 
_{
  "status": "failure",
  "description": "Unauthorized"
}_

I don't what it's happening. Does anyone work with Rainmaker API? Thanks!

Victor

dattatrayhkulkarni-espressif, 2020-12-17:
Hi Victor, 

The Get Node Parameters API is an authenticated API.
After you successfully login, you will get the accesstoken in the response.

For calling the Rainmaker authenticated APIs, you will need to pass this access token as the Authorization  parameter in the HTTP header.

e.g. if you are using Postman as the REST client, you will need to populate the value of Access token in the Authorization HTTP header.

Similarly if you are building some Web apps or Mobile Apps, you will need to use this Access token in the HTTP Authorization 
header.

Hope this addresses your issue.

regards,
dattatray.

[image]

victorK8, 2020-12-17:
Hi dattatray,

I have added Access Token Value to headers in GET request and it is working. 

Thank dattatray!

Regards,
Victor

closed 2020-12-17 by victorK8

bbinet, 2020-12-17:
Hey @victorK8, I'm also interested in an opensource implementation of the rainmaker backend API.
I have planned to work on this, but I haven't found the time to start it yet.

Do you envision to release your implementation of these webservices as opensource software? I may give a hand depending on the language and technologies you've chosen.

victorK8, 2020-12-17:
Hi @bbinet , 

I can't share the code of application because it's for a service from a business. Only I can share that is for an Alexa Skill and it will be developed in Javascript.

Regards

Victor

bbinet, 2020-12-18:
Ok no problem @victorK8