espressif/esp-rainmaker #41: Shows [device connection failed] after QR code scan.

noobGB, 2021-01-07:
Hi,
I am using ESP32, tried FAN and SWITCH example but not able to complete claiming and provisioning step, please check below snapshots, please help me with possible issue here.

Below snapshot shows successful flashing of ESP32 on assisted claiming mode.
[image]

Below shows error on QR code scan,
[image]

Below shows Bluetooth device on manual configuration on app,
[image]

Below show error while trying to connect to BLE device,

[image]

shahpiyushv, 2021-01-07:
@noobGB , what is your phone app version? Some BLE provisioning related issues were fixed in version 2.2.2.

noobGB, 2021-01-07:
Hi Piyush,

App version is v2.2.2, Strange thing is that I was successfully able to add this ESP32 hardware as SWITCH node yesterday then I deleted the node from app just to check FAN example from the same hardware but now I am facing this issue for both FAN and SWITCH example. Does factory reset create some issue as I have tried that as well. Please let me know if you need some more info.

shahpiyushv, 2021-01-07:
@noobGB , none of the events like reboot, reset to factory or erase flash should affect the provisioning. Changing the device type also should not matter as the device type comes into picture after provisioning, not before that. The issue of BLE provisioning failing after a reset to factory was exactly what we had fixed in v2.2.2 of the Android app and I just verified that it is working as expected. I will ask our Android developer to follow up on this.

noobGB, 2021-01-07:
@shahpiyushv , now I tried host driven claiming and then changed the configuration to use softAP instead of BLE for provisioning, now the FAN type node has been added to my user account.
BLE method did not work even after many attempts, will appreciate your response on this after discussion with the android developer.

khushbushah2302, 2021-01-08:
Hi @noobGB ,
From screenshot, it seems like BLE device is disconnecting. I will need android app logs to check this issue.
Can you share android logs ?

noobGB, 2021-01-08:
@khushbushah2302 ,  I figured out that the issue is my system(MOTO G5S PLUS, Android V8.1.0) but this BLE Claim and provisioning for ESP32 worked flawlessly for REDMI 7 Pro (Android V9) and Moto G6(Android V9).
I have installed adb logcat android platform tools, will provide log details if you let me know the adb filtered logcat command for ESP rainmaker logs.

khushbushah2302, 2021-01-11:
Hi @noobGB ,

Here are the few steps to take logs.
1) Connect the device and open ESP RaiMaker. 
2) Clear old logs with `adb logcat -c`
3) Run "`adb shell pidof com.espressif.rainmaker`" command to get pid of ESP RainMaker app .
4) Then, run `adb shell "logcat --pid=12345"` command by replacing 12345 with your pid number (result of first command).

noobGB, 2021-01-12:
@khushbushah2302 , Please check attached log file.

[ESPrainmaker Android logs.txt](https://github.com/espressif/esp-rainmaker/files/5802753/ESPrainmaker.Android.logs.txt)

khushbushah2302, 2021-01-13:
Hi @noobGB ,

Thanks for sharing logs.
I have also MOTO G5S PLUS, Android V8.1.0. Tried on that but app is working fine.
Can you try provisioning device with [ESP BLE ProvisioningApp](https://play.google.com/store/apps/details?id=com.espressif.provble) ?
This app will do only provisioning. I wanted to check if this app is working fine or not with your phone.

noobGB, 2021-01-13:
@khushbushah2302 , tried this app provisioning method, both QR code and manual, shows [Device Disconnected] after the scan. Let me know the logcat commands for this app, I will share log details.

noobGB, 2021-01-17:
@khushbushah2302 , any solution??

khushbushah2302, 2021-01-18:
Hi @noobGB ,

Download [ESP_Prov_BLE.zip](https://github.com/espressif/esp-rainmaker/files/5827991/ESP_Prov_BLE.zip). It contains 2 apks for try.
Install apk and test it in all 3 phones you have.
Let me know the result for both apps.
If any of app will work then we can give app update.

noobGB, 2021-01-18:
@khushbushah2302 , Unfortunately non of the apps worked on my MOTO G5S PLUS, I haven't yet tested these APK files on other phones (REDMI 7 Pro (Android V9) and Moto G6(Android V9)) as ESPrainmaker's Provisioning works flawlessly on these phones.

khushbushah2302, 2021-01-19:
Hi @noobGB ,

Thanks for testing the apps.
No need to check in other phones if app has not worked on Moto G5S Plus phone. It is only needed if any of app has worked on Moto G5S phone.
To debug this issue on your phone, I will need your help. I would like to check with "nRF Connect" app.
Here are the steps:
1. Install ["nRF Connect for Mobile"](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) app.
2. Scan the BLE devices by swiping on "Scanner" tab.
3. If "PROV_XXXXXX" device is available, click on "Connect" for that device.
4. After device connection, app will display "Unknown Service" (check in attached screenshots)
5. Now app will display all the available characteristics and descriptors.
6. Click on "Down" arrow to read the descriptor. (marked by square in screenshot)
7. App should read and display the values of each descriptor. (Values will be "prov-scan", "prov-session", "prov-config" etc.)

[Screenshots.zip](https://github.com/espressif/esp-rainmaker/files/5836185/Screenshots.zip)

Thanks

noobGB, 2021-01-19:
@khushbushah2302 , after {connect}, it does not show anything under {client} tab. Please check snapshots.
[image]
[image]

noobGB, 2021-01-19:
@khushbushah2302 , Don't know exactly what happened here but on above screenshot you could see that the prov device already shows bonded then I use the [Delete bond information] from the nrf connect app. After that ESPrainmaker apps provisioning and claiming process is working fine. I have tried hard reset, wifi reset, erase_flash, all seems to be working fine now on my MOTO G5S PLUS. You may explain what really happened here. BTW, thank you for your support.

khushbushah2302, 2021-02-02:
Hi @noobGB ,

Sorry for late reply. Deleting bond information is like unpairing the device.
Another app is also not able to discover services after device connection and both apps works fine after deleting bond information. So it could be some issue with the BT stack of Android, but I can't be 100% sure.

noobGB, 2021-02-02:
@khushbushah2302 ,

It is strange, as I was able to connect with my Bluetooth speaker even before deleting the bond information. We can consider this issue as resolved for now, will open new issue if somewhat similar issue re-occurs on some other device or android version, thank you for your help and support.

noobGB, 2021-02-02:
@khushbushah2302  or @shahpiyushv , could you please check my comment on this [https://medium.com/the-esp-journal/introducing-esp-rainmaker-8fa968bbb2de] medium article by Kedar, I have asked few commercial questions but did not receive any response.
Please let me know the right channel to ask such questions or directly respond there.

kedars, 2021-02-02:
@noobGB I have responded to your query on the medium article, possibly you missed it, you can check the response here: https://medium.com/p/8fa968bbb2de/info

noobGB, 2021-02-02:
Hi @kedars , I cannot access above provided link, please check.

[image]

kedars, 2021-02-02:
try this one: https://medium.com/@gaurav.gbaba/hi-kedar-da3bfb61ee26#--responses

clients on 'response' icon at the bottom of your post

noobGB, 2021-02-02:
Got it, thank you!!

shahpiyushv, 2021-02-03:
Closing this issue as both threads here seem to be resolved.

closed 2021-02-03 by shahpiyushv