espressif/esp-rainmaker #42: Unable to WiFi provision - WiFi not showing

moosakhalid, 2021-01-09:
Hi Team, 
I'm new to the AWS EduKit. I've been following this tutorial:

https://edukit.workshop.aws/en/getting-started/run-rainmaker.html

Everything goes smooth, until I download the ESP Rainmaker phone app, register an account and try to scan the QR code. The QR code is proper. 
The QR scanner picks up the code but displays an error : "PROV_xxxxxx device not found"

The device does not broadcast any WiFi signal, however does broadcast a bluetooth signal "PROV_xxxxxx" which my phones bluetooth scan picks up but the ESP Rainmaker apps, BLE provisioning scan does not pick up at all. 
No WiFi connection named(PROV_xxxxxx) is picked up from the device by my phones WiFi scan.

I feel like there may be something wrong with the hardware or the ESP app. I simply can't register the device with the ESP RM app this way. 

I think my issue may be different from other issues which report WiFi provisioning. I don't get any errors during build, and the QR code is picked up fine, just the provisioning via ESP app is failing, otherwise the tutorial works fine.

Here's the output returned when uploading the code, you'll notice the alternate URL for getting QR code below too:

> [0;32mI (777) boot: Loaded app from partition at offset 0x20000␛[0m
␛[0;32mI (843) boot: Set actual ota_seq=1 in otadata[0]␛[0m
␛[0;32mI (843) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (844) psram: This chip is ESP32-D0WD␛[0m
␛[0;32mI (848) spiram: Found 64MBit SPI RAM device␛[0m
␛[0;32mI (852) spiram: SPI RAM mode: flash 80m sram 80m␛[0m
␛[0;32mI (857) spiram: PSRAM initialized, cache is in low/high (2-core) mode.␛[0m
␛[0;32mI (864) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (868) cpu_start: Application information:␛[0m
␛[0;32mI (873) cpu_start: Project name:     AWS_IoT_EduKit-Getting_Started␛[0m
␛[0;32mI (880) cpu_start: App version:      d5b92ef␛[0m
␛[0;32mI (885) cpu_start: Compile time:     Jan  8 2021 14:32:23␛[0m
␛[0;32mI (891) cpu_start: ELF file SHA256:  e4cb324775ef3c5f...␛[0m
␛[0;32mI (897) cpu_start: ESP-IDF:          3.40100.200827␛[0m
␛[0;32mI (903) cpu_start: Starting app cpu, entry point is 0x400830b0␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (1406) spiram: SPI SRAM memory test OK␛[0m
␛[0;32mI (1406) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (1407) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM␛[0m
␛[0;32mI (1413) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM␛[0m
␛[0;32mI (1419) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM␛[0m
␛[0;32mI (1425) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM␛[0m
␛[0;32mI (1431) heap_init: At 3FFD32A0 len 0000CD60 (51 KiB): DRAM␛[0m
␛[0;32mI (1438) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (1444) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (1450) heap_init: At 4009E608 len 000019F8 (6 KiB): IRAM␛[0m
␛[0;32mI (1457) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (1462) spiram: Adding pool of 4096K of external SPI memory to heap allocator␛[0m
␛[0;32mI (1482) spi_flash: detected chip: generic␛[0m
␛[0;32mI (1482) spi_flash: flash io: qio␛[0m
␛[0;32mI (1482) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (1491) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations␛[0m
␛[0;32mI (1521) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:3 ␛[0m
␛[0;32mI (1721) ILI9341: Initialization.␛[0m
␛[0;32mI (1921) ILI9341: Display orientation: LANDSCAPE␛[0m
␛[0;32mI (1921) ILI9341: 0x36 command value: 0x08␛[0m
I (1981) wifi:wifi driver task: 3ffd61b4, prio:23, stack:6656, core=0
␛[0;32mI (1981) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
␛[0;32mI (1981) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
I (2001) wifi:wifi firmware version: 3ea4c76
I (2001) wifi:config NVS flash: enabled
I (2001) wifi:config nano formating: disabled
I (2001) wifi:Init dynamic tx buffer num: 32
I (2011) wifi:Init data frame dynamic rx buffer num: 16
I (2011) wifi:Init management frame dynamic rx buffer num: 16
I (2021) wifi:Init management short buffer num: 32
I (2021) wifi:Init static tx buffer num: 8
I (2031) wifi:Init static rx buffer size: 1600
I (2031) wifi:Init static rx buffer num: 4
I (2041) wifi:Init dynamic rx buffer num: 16
␛[0;32mI (2051) esp_claim: Initialising Assisted Claiming. This may take time.␛[0m
␛[0;32mI (2331) esp_claim: Private key already exists. No need to re-initialise it.␛[0m
␛[0;32mI (2411) esp_rmaker_node: Node ID ----- B8F009C5B61C␛[0m
␛[0;32mI (2411) display: configuring the house␛[0m
␛[0;32mI (2411) display: house configured␛[0m
␛[0;32mI (2421) display: lights off␛[0m
␛[0;32mI (3821) display: configuring the temperature␛[0m
␛[0;32mI (3821) display: temperature configured␛[0m
␛[0;32mI (3901) display: configuring the fan␛[0m
␛[0;32mI (3901) display: configured fan_object␛[0m
␛[0;32mI (3901) display: configured fan_strength_slider␛[0m
␛[0;32mI (3901) display: configured fan_sw1␛[0m
␛[0;32mI (3911) display: fan configured␛[0m
␛[0;32mI (3911) esp_rmaker_time_sync: Initializing SNTP. Using the SNTP server: pool.ntp.org␛[0m
␛[0;32mI (3921) esp_rmaker_core: Starting RainMaker Core Task␛[0m
␛[0;32mI (3931) esp_claim: Waiting for assisted claim to finish.␛[0m
␛[0;32mI (3931) wifi_prov_scheme_ble: BT memory released␛[0m
␛[0;32mI (3941) app_wifi: Starting provisioning␛[0m
␛[0;32mI (4071) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0␛[0m
I (4071) wifi:mode : sta (b8:f0:09:c5:b6:1c)
␛[0;32mI (4071) BTDM_INIT: BT controller compile version [219866f]␛[0m
␛[0;32mI (4071) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
␛[0;32mI (4251) protocomm_nimble: BLE Host Task Started␛[0m
␛[0;32mI (4251) wifi_prov_mgr: Provisioning started with service name : PROV_160c3c ␛[0m
␛[0;32mI (4251) app_wifi: Provisioning started␛[0m
␛[0;32mI (4261) app_wifi: Scan this QR code from the phone app for Provisioning.␛[0m
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
                                      
  █▀▀▀▀▀█ ▀▀▀█▄█   ▀▄▄▀▀  █ █▀▀▀▀▀█   
  █ ███ █  ▀▄█ █▄ ▀▄█ ▀█▀▀  █ ███ █   
  █ ▀▀▀ █  ▄▀█▀▄▀ ▀█▄▀██▄▄▀ █ ▀▀▀ █   
  ▀▀▀▀▀▀▀ █▄▀ █▄█▄█ █ █ █ █ ▀▀▀▀▀▀▀   
  ▀█ ▀█ ▀  ▀▄ ▀▄ ▄▀▀▀█  ▀▄▄ ▀▄▄ ▄▄▀   
  █ ▀ ▀ ▀▀█▀  ▀▀▀▀▄▄  ▄  ▀ █  ▄█▄█▀   
  ▄▄▀▄  ▀▀███ ▄▄██▄ ▄▀▄    ▄▀█ ▀▄▄▀   
  ▀███▀▀▀▄█▄█ ▄ ▄█▄▀▀█▄ ▀▄▀▄▄█  ▄     
  █▄ █▄ ▀█▄▀ █▀▀ █▀▄▀▄▀ ▄█  ███▄ ██   
   ▀█ ▄▄▀▀▄ █▄▀▄███▀▄▀█ ▀█ █▀▀ ▀▄▄▀   
  ▄ ▀ ▄▀▀▄█ ▀▀▄▄█▄▀▀█▄█▄█▀▀█ ▀▄ ▄▀    
  █ ██  ▀▀▀███▄  ▀█▄█▄▀▀█▀█ ▄█ ▀▄▄█   
  ▀▀▀▀ ▀▀▀█▄▄▀▀▄▄██▄█▀█ ▀██▀▀▀█▄▄▀    
  █▀▀▀▀▀█  ▀██▀▀▀██ ▄▀▄ █▄█ ▀ █ ▄ ▄   
  █ ███ █ █▄ ▄█▀▀█▀▄█▄▄ ▀██▀▀▀▀▄▄▀▀   
  █ ▀▀▀ █ ▄▄▀▀ ▄█▀█ █▀ ▀▀███▄▀█ █▄█   
  ▀▀▀▀▀▀▀ ▀ ▀▀  ▀▀ ▀     ▀▀▀▀▀▀       
                                      

␛[0;32mI (4461) app_wifi: If QR code is not visible, copy paste the below URL in a browser.

https://rainmaker.espressif.com/qrcode.html?data={"ver":"v1","name":"PROV_160c3c","pop":"5c7007c4","transport":"ble"}

␛[0m
␛[0;32mI (4481) app_wifi: Provisioning Started. Name : PROV_160c3c, POP : 5c7007c4␛[0m
␛[0;32mI (63821) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Temperature":37.25987}}␛[0m
␛[0;32mI (123821) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Temperature":36.68064}}␛[0m
␛[0;32mI (183821) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Temperature":36.37782}}␛[0m
␛[0;32mI (243821) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Temperature":35.71349}}␛[0m
␛[0;32mI (303821) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Temperature":36.31116}}␛[0m
␛[0;32mI (363821) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Temperature":36.10451}}␛[0m
␛[0;32mI (423821) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Temperature":37.63218}}␛[0m
␛[0;32mI (483821) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Temperature":39.31074}}␛[0m
␛[0;32mI (543821) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Temperature":40.48790}}␛[0m
␛[0;32mI (603821) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Temperature":41.08063}}␛[0m
␛[0;32mI (663821) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Temperature":42.05543}}␛[0m
␛[0;32mI (723821) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Temperature":42.60798}}␛[0m

shahpiyushv, 2021-01-11:
@moosakhalid , one small question, what is your phone side OS? If it is Android, we had fixed some BLE provisioning related issues in version 2.2.2. Can you check what your version is?

Moreover, can you enable additional debugging logs on the firmware side from here and retry?

`idf.py menuconfig -> Component config -> Log output -> Default log verbosity -> Debug`

moosakhalid, 2021-01-13:
> @moosakhalid , one small question, what is your phone side OS? If it is Android, we had fixed some BLE provisioning related issues in version 2.2.2. Can you check what your version is?
>
> Moreover, [...]

Hi @shahpiyushv !
Thanks for your response. My Phone's Android version is 10 with kernel 4.14. 117-perf+.

I'm actually flashing the firmware onto the device using PlatformIO extension on Visual Studio Code, so I don't have much idea about using CLI commands for enabling debugging. Also I can't find any idf.py file in the project, perhaps I'm missing it. 
Are there any helpful instructions on enabling debugging on firmware side? The PlatformIO UI on VSCode has a "Verbose upload" button but it's output is way too verbose and only shows C compiler linking and build output.

I hope I'm not confusing you too much here.

shahpiyushv, 2021-01-13:
@moosakhalid , I have not use the VS code + PlatformIO based development setup, but can you check the documentation for menuconfig [here](https://docs.platformio.org/en/latest/frameworks/espidf.html#configuration-for-4-0). That should give you similar options like I have mentioned above.

solroshan, 2021-06-02:
Hi guys, not sure but I have a similar issue for WiFi provisioning from my angle. 
Sorry if is not true space but my story was like that;
I could not find my saved qr code so i deleted the node from ios(v12.5.3) app and respawned my esp with factory reset for new qr code.
I've been working on analog calibrations with Rainmaker for a while. Everything was going on the way till I changed my city also my WiFi network then I needed to reset my network details. Tried to provision with both of BLE and Softapp but ios app returns no-response. Camera cant detect the qr code(screen freezing) Softapp is not working on my ios. Connect button led me to the settings for giving rights for wifi&ble connections to rainmaker but I already give all of permissions to rainmaker🤔
in the end what is the trick of provisioning on when WiFi changes?
Thank you,

solroshan, 2021-06-03:
I fixed my provisioning with BLE provision but i wonder why qr code could not worked after factory reset. Is it an ios app side issue maybe like ios version? Because esp idf monitor did not show any warnings or errors while i tried to provision with qr code.