espressif/esp-rainmaker #43: Rainmaker when WiFi is not available

wmantillar, 2021-01-13:
Hi Team;

I'm new with esp32 but really exited about all possibilities it offers fot IoT Devs, I am using rainmaker for some developments and works great, nevertheless I have an application for a device that need to work in the field where not WiFi is available, so I wonder if there a way to use cellular connection instead (i.e. using # PPP over Serial (PPPoS) client example thought SIM800L or similar), can you give your thoughts about this or any ideas?

shahpiyushv, 2021-01-22:
@wmantillar , sorry for the late reply. Have you checked [this esp-idf example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client)? If it suits your requirement, it may be possible to integrate it with RainMaker. Ofcourse, there could be some challenges, but worth checking out.

wmantillar, 2021-01-22:
> @wmantillar , sorry for the late reply. Have you checked [this esp-idf example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client)? If it suits your requirement, it [...]
@shahpiyushv thank you for your answer, I saw that example, actually my question was aimed at finding out whether or not I have a chance of succeeding trying to integrate that example into Rain Maker or if there is some structural constraint in the Rainmaker stack that will not allow it to do so.  Would you encourage me to explore it?

shahpiyushv, 2021-04-08:
@wmantillar , sorry I missed out on replying here. RainMaker primarily needs some Internet connection to work. The esp-idf  [pppos client example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client) I mentioned above helps create a network interface and so the higher layer application protocols should work seamlessly.

The other important aspect i.e. user node sharing can also be supported, but would require  some efforts in the phone apps as well as the firmware. There are no technical/structural constraints for this, but just some additional work. You may also check out my comment [here](https://github.com/espressif/esp-rainmaker/issues/58#issuecomment-810498440) for some additional info in this context.

Marcarleto, 2023-01-10:
My issue is very similar to wmantillar: I'm developing a project that the device needs to work with or without wi-fi connection. While the code is running and the connection is lost the program still works, but if I start up without the wi-fi I get stuck in the event WIFI_EVENT_STA_DISCONNECTED and printing the message "Disconnected. Connecting to the AP again..." so I can't run my code manually or offline. Can you help me on this?

shahpiyushv, 2023-01-11:
@Marcarleto , is it indeed same as the original issue, which requires internet connectivity even when the same is not available? What exactly are you trying to run, which is nit working?