espressif/esp-rainmaker #44: Relationship about Espressif Cloud Service and Azure Cloud

daluge, 2021-01-19:
Device report data to azure mqtt cloud,  but app device control use cloud service of Espressif? How about their relationship? My team plan to do a wifi dongle to get some products connected to alexa audio. Now, i wonder the bearing and concurrency capability of cloud service of Espressif...

shahpiyushv, 2021-01-20:
@daluge, the query isn't really clear to me. Do you mean to say that your device is using azure IoT and you want to interface it with ESP RainMaker? If that is not the case, can you elaborate more on the requirement here?

iz0eyj, 2021-01-21:
@shahpiyushv I think he is asking if the servers on which the service is based are on Azure and, if so, what relationship there are between Espressif and Microsoft.

shahpiyushv, 2021-01-21:
@iz0eyj , thanks for helping out here. @daluge , ESP RainMaker is deployed on Amazon Web Services (AWS) using various serverless components. The devices connect to the AWS IoT Core service (which is an MQTT broker) whereas clients like phone apps communicate over REST APIS via the AWS API Gateway service. There are many other AWS services involved which tie all of this together. If you want to know more about why we chose the serverless approach, you can read about it in this [blog post](https://medium.com/the-esp-journal/esp-rainmaker-and-serverless-d144d8a71987)

daluge, 2021-01-22:
@shahpiyushv  thanks for answer!!! sorry for my error despcription before, it's amazon cloud, not azure cloud... and blog post do help me.

shahpiyushv, 2021-01-22:
@daluge , closing this issue since I guess your query has been answered. Please feel free to ask any more questions you may have.

closed 2021-01-22 by shahpiyushv

daluge, 2021-01-25:
@shahpiyushv  thanks!