espressif/esp-rainmaker #45: How to combine ESP32-Rainmaker and ESP32-ADF?

daluge, 2021-01-25:
ESP32-ADF use idf version v3.3.2, but ESP32-Rainmaker usb idf version v4.xx.  
I checkout higher version of idf for ESP32-ADF, it compile fail.
Then I also checkout lower version(v3.3.2) for ESP32-Rainmaker, it compile fail too.
My project may need to play mp3 or tone except interaction through amazon alexa using ESP32-Rainmaker.

shahpiyushv, 2021-01-25:
@daluge , can you please refer to [this issue](https://github.com/espressif/esp-adf/issues/563) in esp-adf and see if it helps?

daluge, 2021-01-26:
@shahpiyushv thx. I rewrite CMakeLists.txt in esp-adf directory that use the same idf as esp-rainmaker use, and now it succeed to compile for a project in esp-adf. 
However, i still bother how to use a project in esp-adf as main project, and use esp-rainmaker as component? 
I tryed to rewrite CMakeList.txt in project "play_mp3", but when i invoke function in esp_rainmaker, then it failed to compile. 

Error info like:
/home/xxx/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/xxx/esp/project-adf/esp-rainmaker/examples/test_desktop/main/app_main.c:79: undefined reference to `esp_rmaker_device_add_param'

My project directory as follows:
project-adf:
---> esp-rainmaker
---> play_mp3

daluge, 2021-01-26:
I have succeed to compile and run the project with adf and rainmaker.
let me do more test and functions ...

daluge, 2021-02-04:
1. sdk change:
   perpare esp-adf with the latest version.
   perpare esp-rainmaker with the latest version.
   perpare esp-idf, change its version to v4.1.
   modify file "esp-adf/CMakeLists.txt", change IDF_PATH -> set(ENV{IDF_PATH} "$ENV{esp-idf}/")
   cuz esp-adf has its own esp-idf which version is v3.3.2, and for convinence to use 2 sdk, here i use esp-idf outside.
  
2. check an example in esp-adf can compile and work,
   check an example in esp-rainmaker...
   
3. then new a directory 'project-esp-rainmaker' in the same directory of 'esp-adf' and 'esp-idf',
   copy the example project like 'pipeline_http_mp3' in esp-adf to directory 'project-esp-rainmaker',
   copy the whole directory 'esp-rainmaker' to directory 'project-esp-rainmaker/pipeline_http_mp3',
   
4. dependent (project) change:
   change file 'project-esp-rainmaker/pipeline_http_mp3/CMakeLists.txt', 
   add code after 'include($ENV{ADF_PATH}/CMakeLists.txt)'
   set(EXTRA_COMPONENT_DIRS 	$ENV{ADF_PATH}/components
							$ENV{ADF_PATH}/../project-esp-rainmaker/pipeline_http_mp3/esp-rainmaker/components
							$ENV{ADF_PATH}/../project-esp-rainmaker/pipeline_http_mp3/esp-rainmaker/examples/common
   )
   
5. dependent (content) change:
   change file 'project-esp-rainmaker/pipeline_http_mp3/main/CMakeLists.txt', 
   add .c and .h from 'pipeline_http_mp3/esp-rainmaker'. 
   
6. component change:
   change file 'project-esp-rainmaker/pipeline_http_mp3/main/component.mk', if xx.mp3 used.
   
7. menuconfig setup:
   -1- Partition Talbe ---> Custom partition table CSV
       copy partitions.csv in 'project-rainmaker\pipeline_http_mp3\esp-rainmaker\examples\switch' 
	   to 'project-esp-rainmaker/pipeline_http_mp3'
	   
8. project change:
   -1- use one net config:
       use net config of esp-rainmaker, delete net config code of esp-adf if used.
   -2- use one storage:
       use storage if esp-rainmaker.
   -3- peripheral and io:
       comment code about app_driver.c of esp-rainmaker,
	   comment code about audio_board_key_init() of esp-adf.
	   only use either or write by myself.
   
project directory:
--> esp-adf
--> esp-idf
--> project-rainmaker
    --> pipeline_http_mp3
	    --> components
		--> esp-rainmaker
		--> main