espressif/esp-rainmaker #47: Enhancement proposal (Switch - Socket, Door, Humidity, etc) Icons in phone Apps

ilijamr, 2021-02-16:
Beside this:
https://github.com/espressif/esp-rainmaker/issues/7
there should be a permanent switch UI element whilst current switch element should be renamed to a socket or wall socket.
Door UI element should have Open-Closed and Battery state.
Humidity UI element should have Relative Humidity in percentage.
In addition, just to name a few more useful UI elements:
CO, CO2, water leak, shock, sensors (true - false)
Wind direction and strength
Rain sensor (true-false)
Day / Night sensor (On-Off)
Light intensity (lm, lx, cd)
Human Body sensor

shahpiyushv, 2021-02-16:
@ilijamr , thanks for your suggestions. These would definitely be useful. We will take up this task as soon as we finish off with the current pending features.

jberezin, 2021-02-22:
In the case of water leak and other sensors, it would be nice if third party integration supported alarms for conditions such as water leak or temperature outside of range for the Alexa echo to say "water leak in basement!" without needing a voice prompt.