espressif/esp-rainmaker #48: how to combine ESP-IDF with ESP Rainmaker components

alexiskyriakou, 2021-02-18:
Hello,

does anyone knows how can i combine ESP-idf and ESP Rainmaker components? 

i had open GPIO example from Rainmaker directory  and add in c_cpp_properties.json the following:

>{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "${workspaceFolder}/**",
                "${env:IDF_PATH}/components/**",
                "C:/Users/Alexis/esp/esp-rainmaker/components/**",
                "C:/Users/Alexis/esp/esp-rainmaker/components/esp_rainmaker/include/**",
                "C:/Users/Alexis/esp/esp-rainmaker/examples/common/**",
                "C:/Users/Alexis/esp/esp-idf/components/**",
                "C:/Users/Alexis/esp/esp-idf/components/freertos/**",
                "C:/Users/Alexis/esp/esp-idf/components/freertos/include/freertos/**"
            ],
            "defines": [
                "_DEBUG",
                "UNICODE",
                "_UNICODE"
            ],
            "cStandard": "c17",
            "cppStandard": "c++17",
            "intelliSenseMode": "windows-msvc-x64"
        }
    ],
    "version": 4
}

and gives me erros that it can not find stddef.h, limit.h, stdint.h in almost all included files

I am using Visual Studio Code ESP-IDF extension

Thank you
Alexis

ameyinamdar, 2021-02-18:
Hi @alexiskyriakou You don't need to change any file. Once the IDF is configured properly in Espressif VS Code plugin, you can simply open any of the application in esp-rainmaker/examples/ folder and directly compile through VS Code plugin. The asp-rainmaker components are added by the application's CMakelist.txt and other IDF components are added by IDF build system.

alexiskyriakou, 2021-02-18:
I believe that idf configured correct in VS code because i can open, bulid and flash examples under esp-idf.

But when i open an example from esp rainmaker, VScode does not recognize the include files.. 

Both esp-idf and esp rainmaker are in the same folder like:
 esp -> esp-idf
        -> esp-rainmaker

ameyinamdar, 2021-02-19:
@alexiskyriakou VSCode not recognising the include files is probably just navigation issue. Can you compile the rainmaker example in this configuration? If not, can you please provide the log?

alexiskyriakou, 2021-02-19:
No i can not compile.

Here is the log file for Fan Example [CMakeOutput.log](https://github.com/espressif/esp-rainmaker/files/6008125/CMakeOutput.log)

alexiskyriakou, 2021-02-21:
Finally i fixed it... I clone the esp rainmaker repository and now builds flash and monitor rainmaker examples...

alexiskyriakou, 2021-02-22:
i have one more question...

I create a project under EspRainmaker directory and works fine. My question is which is the proper way to move that project under Documents folders ? I want to do this because my Documents folder is sync to cloud for backup purposes

Thank you in advance

ameyinamdar, 2021-02-22:
You need to copy components/ folder from asp-rainmaker and adjust relative paths in application's CMakeLists.txt

```
# Add RainMaker components and other common application components
set(EXTRA_COMPONENT_DIRS ${CMAKE_CURRENT_LIST_DIR}/../../components ${CMAKE_CURRENT_LIST_DIR}/../common)
```

alexiskyriakou, 2021-02-24:
Thanks @ameyinamdar its woking now.

closed 2021-02-24 by alexiskyriakou