espressif/esp-rainmaker #50: Sharing nodes using API or CLI is not working any more

wmantillar, 2021-02-27:
Dear Team;

I am using rainmaker APIs for a development, but until some time ago I could share my nodes just using "PUT​/{version}​/user​/nodes​/sharing" and putting the email of the user to share to, nevertheless I cannot do that any more, even when I see "success" it returns the following message, but the other user is not able to see the nodes, seems like it only creates a request, but a new step is required, can you please advice me what else should I do?

{
  "status": "success",
  "description": "Node sharing request added for the user successfully",
  "request_id": "8VA7XKjf645v6uyomZExAD"
}

shahpiyushv, 2021-02-27:
@wmantillar, sorry for the inconvenience. The work flow for node sharing has been changed such that the secondary user needs to explicitly accept the request for the nodes to show up under his/her account. The CLI will be updated next week and this will soon be available in the phone apps as well.

wmantillar, 2021-02-27:
@shahpiyushv thankyou for the answer; can you please advise which API should I use in the secondary user to accept the request?

shahpiyushv, 2021-04-08:
@wmantillar , the CLI has now been updated. Some information of the new node  sharing workflow can be found [here](https://rainmaker.espressif.com/docs/node-sharing.html#workflow). The [Swagger API documentation](https://swaggerapis.rainmaker.espressif.com/) has also been updated appropriately.

closed 2021-04-08 by shahpiyushv