espressif/esp-rainmaker #51: ESP32S2 Saola 1 refuses to get esp claim when running the switch example

nell-du, 2021-03-02:
## Board
ESP32S2 Saola 1

## ESP-IDF
Head of the repository

## ESP-RainMaker
Head of the repository with all the submodules updated to the latest 

## Description
I was simply following the startup instructions for the rainmaker: 
https://rainmaker.espressif.com/docs/get-started.html

Here is the log information:
I (87170) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (88620) app_wifi: Connected with IP Address:192.168.5.131
I (88620) esp_netif_handlers: sta ip: 192.168.5.131, mask: 255.255.255.0, gw: 192.168.5.1
I (88620) wifi_prov_mgr: STA Got IP
I (88630) app_wifi: Provisioning successful
I (88630) app_main: RainMaker Claim Started.
I (88640) esp_claim: Starting the Self Claim Process. This may take time.
I (90010) wifi:station: ee:90:33:72:13:39 leave, AID = 1, bss_flags is 134259, bss:0x3ffe5288
I (90010) wifi:new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1
W (90010) wifi:<ba-del>idx
I (90010) wifi:mode : sta (7c:df:a1:08:e2:a0)
I (90020) wifi_prov_mgr: Provisioning stopped
**W (98430) esp-tls: Failed to open new connection in specified timeout
E (98430) TRANS_SSL: Failed to open a new connection
E (98430) HTTP_CLIENT: Connection failed, sock < 0
E (98430) esp_claim: Failed to open connection to https://esp-claiming.rainmaker.espressif.com/claim/initiate
E (98440) esp_claim: Claim Init Request Failed.
E (98450) esp_claim: Claim Init Sequence Failed.
I (98450) app_main: RainMaker Claim Failed.
E (98460) esp_rmaker_core: esp_rmaker_self_claim_perform() returned -1. Aborting
W (98470) esp_rmaker_mqtt: esp_rmaker_mqtt_publish not registered**
I (153230) esp_rmaker_time: SNTP Synchronised.
I (153230) esp_rmaker_time: The current time is: Tue Mar  2 16:21:59 2021 +0800[CST], DST: No.

Here is the log from Wireshark when provisioning and trying to connect to the rainmakers server:
[esp-dumbs.pcapng.gz](https://github.com/espressif/esp-rainmaker/files/6067243/esp-dumbs.pcapng.gz)
[image]

shahpiyushv, 2021-03-02:
@nell-du , we have observed some DNS Issues for espressif.com since the last few hours and that could probably be the reason for this. Can you just try this again. Please also check if you can ping `esp-claiming.rainmaker.espressif.com` from a host machine connected to the same network.

nell-du, 2021-03-03:
Ya we did do so yesterday (the same day when I posted this issue) we were struggling to get rainmaker.espressif.com connected consistently. Dig also sometimes gave us results and sometimes didn't. The device just received the certificate. Thank you.

closed 2021-03-03 by nell-du