espressif/esp-rainmaker #53: Multi-Device Example crash on posting temperature data

JohnMacrae, 2021-03-03:
ESP32-S2

To reproduce:
Compile example and upload. 

Wait for Temperature to be updated

Log:

I (32458) esp_rmaker_time: The current time is: Thu Mar  4 01:40:12 2021 +0800[CST], DST: No.
I (60508) esp_rmaker_param: Reporting params: {"Temperature Sensor":{"Temperature":25.50000}}
E (65498) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (65498) task_wdt:  - IDLE (CPU 0)
E (65498) task_wdt: Tasks currently running:
E (65498) task_wdt: CPU 0: esp_timer
E (65498) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400C15A3:0x3FFC81B0 0x400245B2:0x3FFC81D0 0x4010CBB4:0x3FF9F7F0  |<-CORRUPTED
0x400c15a3: task_wdt_isr at C:/esp/esp-idf/components/esp_common/src/task_wdt.c:189

0x400245b2: _xt_lowint1 at C:/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4010cbb4: aes_ll_get_state at C:\esp\esp-idf\esp-rainmaker\examples\multi_device\build/../../../../components/hal/esp32s2/include/hal/aes_ll.h:140 (discriminator 1)
 (inlined by) aes_hal_wait_done at C:/esp/esp-idf/components/hal/aes_hal.c:92 (discriminator 1)

Thank you.

JohnMacrae, 2021-03-04:
I decided to do some further testing and this problem is only on ESP32S2. On the C3, it runs well.

shahpiyushv, 2021-04-08:
@JohnMacrae , some fixes related to this were made in [this MR](https://github.com/espressif/esp-rainmaker/commit/0fba68bbf7effefbc7f2348ce815a4b528f7a6f5). Can you confirm if it works for you?

JohnMacrae, 2021-04-08:
@shahpiyushv many thanks for this MR. Seems to be working well on both S2 and C3 over the last few hours :)

closed 2021-04-08 by JohnMacrae

JohnMacrae, 2021-04-09:
All running well - many thanks for this MR.