espressif/esp-rainmaker #54: Questions : About time zone and temperature value

alexiskyriakou, 2021-03-12:
Hello guys,

According to the title i would like to ask how can i set the time and time zone with the mobile phone to get the local time automatically?

And how can i make the value of temperature to be only an integer value, just 22, not a float like in the picture below?

[image]

shahpiyushv, 2021-03-12:
@alexiskyriakou , for timezone, please check out the docs [here](https://rainmaker.espressif.com/docs/time-service.html). The simplest option is to set the timezone using the `CONFIG_ESP_RMAKER_DEF_TIMEZONE` config option. Going ahead, the phone apps will allow setting the timezone by using the [timezone service](https://rainmaker.espressif.com/docs/time-service.html#rainmaker-service).

WIth regards to the temperature values, to change it to integer, you can edit [this function](https://github.com/espressif/esp-rainmaker/blob/38a46da76098714c3e3bfbe230d6ea00f656c6e5/components/esp_rainmaker/src/standard_types/esp_rmaker_standard_params.c#L114) directly (accept `int` value as argument and then use `esp_rmaker_int(val)`) or create a copy of this function in your own example and use this.

We will also check how to better render such values in the phone app. Float values often get tricky due to precision issues, but we can truncate such values (say 22.9) or adjust the size such that it shows up on a single line.

alexiskyriakou, 2021-03-23:
Hello @shahpiyushv and sorry for the late reply.

I know that is easier to set the time zone in _CONFIG_ESP_RMAKER_DEF_TIMEZONE_ but if I want to get the time zone from phone app how can I perform that? where can I find any example/instructions on how to use the _esp_rmaker_timezone_service_enable(void);_ ?

Thank you

shahpiyushv, 2021-04-05:
@alexiskyriakou , sorry for the late reply. Somehow, I missed this comment. Details of the `esp_rmaker_timezone_service_enable()` API can be found [here](https://rainmaker.espressif.com/docs/time-service.html#rainmaker-service). However, this is not enabled in the phone apps yet and can be used only from the RainMaker CLI.

alexiskyriakou, 2021-04-06:
@shahpiyushv thank you

shahpiyushv, 2021-12-03:
Setting timezone is now available in the phone apps from node details screen. Timezone also gets automatically set during the last step of provisioning.

closed 2021-12-03 by shahpiyushv