espressif/esp-rainmaker #56: Is there a way to preset temperature and humidity for climate control?

Asteliks, 2021-03-15:
Basically as stated in the title. I want to make an ESP32 based climate control device that controls a heater and a humidifier with a PID algorithm. The question is dose rainmaker support inputting humidity and temperature values?

kedars, 2021-03-16:
Yes, you can add humidity and temperature parameters in the RainMaker's data model. [This](https://github.com/espressif/esp-rainmaker/blob/master/examples/switch/main/app_main.c#L151) is how you can add parameters to your device.

Asteliks, 2021-03-22:
Thanks didn't see that

Asteliks, 2021-03-27:
I've tried reusing the code you've mentioned but there is no easy value input in the rainmaker app. TBH the temperature value is stuck at the default value and there is no changing of it. I've also tried going through the documentation several times but no luck there. Basically I need a simple parameter that will allow me to input the wanted temperatura and humidity values

kedars, 2021-03-31:
The temperature sensor, being a sensor, the value is read-only from the phone-app, but modifiable from the device. I am not sure what exactly you are looking for, one of the following options?
* In case you want a read-write temperature value (like a target temperature), that can be modified from the phone app, then you can add an integer parameter that is read-write. You can refer to this: https://github.com/espressif/esp-rainmaker/blob/master/examples/gpio/main/app_main.c#L74
* If you want a read-only temperature value, but want to modify it from the firmware you can refer to this: https://github.com/espressif/esp-rainmaker/blob/master/examples/temperature_sensor/main/app_driver.c#L51