espressif/esp-rainmaker #58: Adding an option to bypass WiFi

Maldus512, 2021-03-22:
I am developing a solution that can use either WiFi or Ethernet connectivity. Unfortunately the rainmaker main task is dependent on WiFi connection and waits for one before doing anything (https://github.com/espressif/esp-rainmaker/blob/37b6a24dfa22303dcf591c121d95ae80af0bb1ef/components/esp_rainmaker/src/core/esp_rmaker_core.c#L243).
For now I'll probably add some workaround in a fork, but I think that having the device connected via Ethernet is not so farfetched. There should be an API that allows to skip the WiFi check.

shahpiyushv, 2021-03-30:
Adding link to the [forum post](https://www.esp32.com/viewtopic.php?f=41&t=19742&sid=3263f6e6220364e72d37e1833cc259b8) for some context. Also copying the relevant comment here:

*At the specification level, [user node mapping](https://rainmaker.espressif.com/docs/user-node-mapping.html) is independent of Wi-Fi provisioning. The [esp_rmaker_start_user_node_mapping()](https://docs.espressif.com/projects/esp-rainmaker/en/latest/c-api-reference/rainmaker_core.html#_CPPv434esp_rmaker_start_user_node_mappingPcPc) API can be used on the device side to trigger the device side workflow. Even on the phone app side, you just need to invoke the [add user-node mapping](https://swaggerapis.rainmaker.espressif.com/#/User%20Node%20Association/addRemoveUserNodeMapping) REST API. All the required support is already available in the apps, but since this isn't the usual workflow, as of now, you will have to make your own customisations in our [Android](https://github.com/espressif/esp-rainmaker-android)/[iOS](https://github.com/espressif/esp-rainmaker-ios) source code. We can help out with information regarding actual APIs to be called and their ordering if you want to take that path.*

Of course, an additional change w.r.t the event handling will also be required for Ethernet connections.

bbinet, 2022-02-01:
+1 for adding an option to bypass WiFi: this would also be useful when using 4G modem instead of ethernet connectivity (see: #107)

Maldus512, 2024-02-27:
I need to come back to this because it's still an issue. Now I have a custom application that initiates the node mapping, but the procedure does not work (the mapping request times out on the server) unless the node has been claimed (I'm using self claiming), and that part is locked behind a WiFi connection. However, my device has also an ethernet interface.

Would you be willing to accept a PR? I could add an API to bypass WiFi or listen for ethernet events as well.

chirag-77, 2025-11-25:
is there any update on this, to use LAN instead of wifi?