espressif/esp-rainmaker #64: Buffer size 2048 not sufficient for node config

Maldus512, 2021-04-21:
Hello,
During a device provisioning with the default IOS ESP-Rainmaker application I encountered an issue at the node association step. Once the device was connected to a WiFi network the association failed with the message in the title, printed by line https://github.com/espressif/esp-rainmaker/blob/ec23bd8aa6843966fd13e92940e41863779260d4/components/esp_rainmaker/src/core/esp_rmaker_node_config.c#L233

I immediately fixed the problem by setting `CONFIG_ESP_RMAKER_MAX_NODE_CONFIG_SIZE` to 4096 (instead of 2048). I apologize for not showing the exact logs, but I did not save them in the moment and I could not reproduce the issue.

Is the default value (2048) not enough in certain cases? I usually test the application on an Android smartphone and never had this issue. Are there other values that I should increase for safety?

shahpiyushv, 2021-04-21:
@Maldus512 , if you use the multi_device example, and add any additional service, like say timezone service, the node config indeed goes beyond 2048. Something similar may have happened in your case.

Maldus512, 2021-04-22:
Thank you, and sorry again for the fuzzy description. I'm not sure yet of what I could have added that makes the configuration bigger, but I'll be mindful of this possibility from now on.

closed 2021-04-22 by Maldus512

ajarsco, 2021-07-25:
I also encounter the same issue! How to solve this. Happened during adding multiple switches...
error:

⸮E (8799) esp_rmaker_node config: Buffer size 2048 not sufficient for Node Config.
 E (8800) esp_rmaker_node_config: Could not get node configuration for reporting to cloud 
E (8803) esp_rmaker_core: Report node config failed. E (8808) esp_rmaker_core: Aborting!!!

ajarsco, 2021-07-25:
> Thank you, and sorry again for the fuzzy description. I'm not sure yet of what I could have added that makes the configuration bigger, but I'll be mindful of this possibility from now on.

> @Maldus512 , if you use the multi_device example, and add any additional service, like say timezone service, the node config indeed goes beyond 2048. Something similar may have happened in your case.

> Thank you, and sorry again for the fuzzy description. I'm not sure yet of what I could have added that makes the configuration bigger, but I'll be mindful of this possibility from now on.

> @Maldus512 , if you use the multi_device example, and add any additional service, like say timezone service, the node config indeed goes beyond 2048. Something similar may have happened in your case.

I also encounter the same issue! How to solve this. Happened during adding multiple switches...
error:

⸮E (8799) esp_rmaker_node config: Buffer size 2048 not sufficient for Node Config.
E (8800) esp_rmaker_node_config: Could not get node configuration for reporting to cloud
E (8803) esp_rmaker_core: Report node config failed. E (8808) esp_rmaker_core: Aborting!!!

shahpiyushv, 2021-07-25:
@ajarsco The logic was changed some time back such that the buffer would get allocated dynamically, instead of limiting to the config option (default being 2048). Can you just pull the latest esp-rainmaker code and retry?

ajarsco, 2021-07-25:
Yeah I have changed the default 2048 to 4096..
Now that issue is gone.. but the device is not connecting to the app..

Can you give me an example for multiple switches in arduino ide

shahpiyushv, 2021-07-26:
@ajarsco as mentioned above, the logic itself has changed and the `CONFIG_ESP_RMAKER_MAX_NODE_CONFIG_SIZE` no more exists. Even if this is currently working for you, updating to the latest esp-rainmaker is recommended. 

Can you elaborate more on the point `the device is not connecting to the app..`? What exactly is failing?

For multiple switches, basically you just need to define one more switch like this

```
static int gpio_switch2 = 17;
static Switch my_switch2("Switch2", &gpio_switch2);
```

Add a callback and add it to the node

```
    //Standard switch device
    my_switch2.addCb(write_callback);
    
    //Add switch device to the node   
    my_node.addDevice(my_switch2);
```

And in the [write_callback](https://github.com/espressif/arduino-esp32/blob/6e47e18a1b6f18d5f0b9b21eba2a202ee30601ef/libraries/RainMaker/examples/RMakerSwitch/RMakerSwitch.ino#L35), just check for the device name in addition to param name.

You can also check the older Arduino release, which had a switch and a custom device (Fan) in the same node. Check that code [here](https://github.com/espressif/arduino-esp32/blob/2.0.0-alph).