espressif/esp-rainmaker #65: Impossible to delete first/last node

JohnMacrae, 2021-04-21:
There seems to be no way of deleting the first node created on the dashboard.  Deleting from the phone app still leaves the node listed. There is no button on the dashboard to remove the last instance.

shahpiyushv, 2021-04-22:
@JohnMacrae , the nodes you see on dashboard and the ones you see in the phone app are actually "admin" and "end user" views respectively. End user association can be removed, but the admin user association with a node id stays for lifetime. Please check out the docs here to understand these user-roles.

Note that if some other user claims the same node (basically the same mac address), a new node id gets assigned for that user:mac combination. The old entry for the old user will still stay as it is.

JohnMacrae, 2021-04-22:
OK - I understand - many thanks for explaining :)

shahpiyushv, 2021-05-19:
Closing this issue since it has been answered.

closed 2021-05-19 by shahpiyushv