espressif/esp-rainmaker #66: Dashboard: Cannot delete OTA job once triggered

JohnMacrae, 2021-04-21:
Is there a method to do this? Otherwise failed tests just leave a large list of unfulfilled jobs.

shahpiyushv, 2021-04-22:
@JohnMacrae , this is actually on purpose, so that historical data does not get deleted inadvertently. However, we could still add an option to first "archive" a job and then allow deleting only archived jobs. This will help have a clean view of the jobs table, yet address the original problem I mentioned. We will take this up as an enhancement in future,

JohnMacrae, 2021-04-22:
Thanks @shahpiyushv :)

CodorStelian, 2021-09-07:
Also, a clear option for all node logs like errors should be implemented. Is annoying to see hundreds of errors from previous versions.
Thanks

shahpiyushv, 2021-09-07:
@CodorStelian , Noted!

shahpiyushv, 2021-09-13:
@JohnMacrae . closing this issue since the RainMaker dashboard now gives an option to archive OTA jobs. Please have a look.

closed 2021-09-13 by shahpiyushv