espressif/esp-rainmaker #68: Need help to configure switch example with 2 channel relay

solroshan, 2021-05-16:
Hello esp lovers and those who give their hearts;

I'm trying to use rainmaker with;
2 channel relay module with 2 buttons,
DHT11 Temp&humy sensor,
Capacitive soil sensor,
and a Fan device.

First of all I tried to multiply switch example for 2 channel relay module but at last my phone app's 2 buttons controling both of relay channels at same time.
I did not try dht, hygro and fan devices yet but im not sure to use dht11 and capacitive soil sensors with rainmaker. Can you enlight me how can I fix my switch example for 2 ch relay and is rainmaker suitable for dht11 and capacitive soil sensor? 

This is my app_priv.h, app_driver.c and app_main.c details in txt.
[priv_driver_main_codes.txt](https://github.com/espressif/esp-rainmaker/files/6489740/priv_driver_main_codes.txt)

Thanks for helps.

shahpiyushv, 2021-05-17:
@solroshan 

> First of all I tried to multiply switch example for 2 channel relay module but at last my phone app's 2 buttons controling both of relay channels at same time.

The reason for this lies in your write_cb. You have registered the same callback for both your switch devices and the write_cb isn't checking the device name to decide whether to  call `app_driver_set_water_state(val.val.b)` or `app_driver_set_light_state(val.val.b)`. You can check the below line in multi_device example as a reference for such checks:
 https://github.com/espressif/esp-rainmaker/blob/ec23bd8aa6843966fd13e92940e41863779260d4/examples/multi_device/main/app_main.c#L44

> I did not try dht, hygro and fan devices yet but im not sure to use dht11 and capacitive soil sensors with rainmaker. Can you enlight me how can I fix my switch example for 2 ch relay and is rainmaker suitable for dht11 and capacitive soil sensor?

ESP RainMaker is just a framework to allow reporting or controlling any type of parameters. It is independent of the actual hardware or drivers. If your use case can map the sensor data to a RainMaker parameter (of type boolean, integer, float or string), you should be good to go. As you can see, the [temperature_sensor](https://github.com/espressif/esp-rainmaker/tree/master/examples/temperature_sensor) example reports dummy sensor data. You can use it as a reference and report the actual data that you read from your sensor.

solroshan, 2021-05-17:
> @solroshan
>
>
> The reason for this lies in your write_cb. You have registered the same callback for both your switch devices and the write_cb isn't checking the device name to decide whether to call [...]

I'll make corrections immediately and I'll come back to share the main file of the 2ch relay (switch) that works as it should be under this issue topic.

Thanks for help @shahpiyushv

solroshan, 2021-05-18:
I fixed 2 ch relay part with @shahpiyushv precious helps and here my latest write_cb lines they are;

`
    
    //callback for ch1    

    static esp_err_t ch1_write_cb(const esp_rmaker_device_t *ch1_device, const esp_rmaker_param_t *param,
            const esp_rmaker_param_val_t val, void *priv_data, esp_rmaker_write_ctx_t *ctx)
            {
    if (ctx) {
        ESP_LOGI(TAG, "Received write request via : %s", esp_rmaker_device_cb_src_to_str(ctx->src));
    }
    const char *ch1_device_name = esp_rmaker_device_get_name(ch1_device);
    const char *param_name = esp_rmaker_param_get_name(param);

    if (strcmp(esp_rmaker_param_get_name(param), ESP_RMAKER_DEF_POWER_NAME) == 0) {
        ESP_LOGI(TAG, "Received value = %s for %s - %s",
                val.val.b? "true" : "false", ch1_device_name, param_name);
        if (strcmp(ch1_device_name, "Ch1") == 0) {
        app_driver_set_ch1_state(val.val.b);
    }
    } 
    esp_rmaker_param_update_and_report(param, val);
    return ESP_OK;

    }
    //callback for ch2

    static esp_err_t ch2_write_cb(const esp_rmaker_device_t *ch2_device, const esp_rmaker_param_t *param,
    const esp_rmaker_param_val_t val, void *priv_data, esp_rmaker_write_ctx_t *ctx)
    {
    if (ctx) {
    ESP_LOGI(TAG, "Received write request via : %s", esp_rmaker_device_cb_src_to_str(ctx->src));
    }
    const char *ch2_device_name = esp_rmaker_device_get_name(ch2_device);
    const char *param_name = esp_rmaker_param_get_name(param);

    if (strcmp(esp_rmaker_param_get_name(param), ESP_RMAKER_DEF_POWER_NAME) == 0) {
        ESP_LOGI(TAG, "Received value = %s for %s - %s",
                val.val.b? "true" : "false", ch2_device_name, param_name);
        if (strcmp(ch2_device_name, "Işık") == 0) {
        app_driver_set_ch2_state(val.val.b);
    }
    } 
    esp_rmaker_param_update_and_report(param, val);
    return ESP_OK;
    }`
and this lines under void app_main()
`   esp_rmaker_device_add_cb(switch_device_water, water_write_cb, NULL);
    esp_rmaker_device_add_cb(switch_device_light, light_write_cb, NULL);`

may the force be with you @shahpiyushv

closed 2021-05-18 by solroshan

shahpiyushv, 2021-05-18:
@solroshan , good to know that this is resolved. However, you can even have a common write callback and inside that, just check device name using `esp_rmaker_device_get_name()` to decide whether to call `app_driver_set_ch1_state()` or `app_driver_set_ch2_state()`.

solroshan, 2021-05-19:
> @solroshan , good to know that this is resolved. However, you can even have a common write callback and inside that, just check device name using `esp_rmaker_device_get_name()` to decide whether to call `app_driver_set_ch1_state()` or `app_driver_set_ch2_state()`.

Thanks much @shahpiyushv 🤟🏿
Now I arranged them inside common write callback lines.

So would you suggest I focus on the [MCPWM BLDC control example](https://github.com/espressif/esp-afr-sdk/blob/6abb2ca3350d1f41659069239ed62811e466ffb1/examples/peripherals/mcpwm/mcpwm_bldc_control/main/mcpwm_bldc_control_hall_sensor_example.c) to be able to use the fan device(its 5v brushless mini fan device)
and [ADC example](https://github.com/espressif/ESP8266_RTOS_SDK/blob/d45071563cebe9ca520cbed2537dc840b4d6a1e6/examples/peripherals/adc/main/adc_example_main.c) for capacitive soil sensor?

Maybe the road is rough for me at this days but I believe it will be smooth on the way back 👾

shahpiyushv, 2021-05-19:
@solroshan , frankly speaking, I am not much aware of the hardware interfacing or the pointers you shared. RainMaker itself is independent of the hardware/drivers. So, if the above can work fine for you with esp-idf, they should be good enough to integrate with RainMaker.

solroshan, 2021-05-19:
So, if the above can work fine for you with esp-idf, they should be good enough to integrate with RainMaker.

That sounds good @shahpiyushv and sir, I am grateful for your kindful helps. 🙌🏿

shahpiyushv, 2021-06-24:
@adukale , I don't know rest of your code, like how you have initialised switch2_device and all, but why is your callback so complex? Any reason why you report false when you get true and vice versa? For a normal 2 channel plug, the callback should ideally have been as simple as this:

```c
static esp_err_t write_cb(const esp_rmaker_device_t *device, const esp_rmaker_param_t *param,
        const esp_rmaker_param_val_t val, void *priv_data, esp_rmaker_write_ctx_t *ctx)
{
    if (ctx) {
        ESP_LOGI(TAG, "Received write request via : %s", esp_rmaker_device_cb_src_to_str(ctx->src));
    }
    if (strcmp(esp_rmaker_param_get_name(param), ESP_RMAKER_DEF_POWER_NAME) == 0) {
        ESP_LOGI(TAG, "Received value = %s for %s - %s",
                val.val.b? "true" : "false", esp_rmaker_device_get_name(device),
                esp_rmaker_param_get_name(param));
        if (strcmp(esp_rmaker_device_get_name(device),"Switch1")==0) {
            app_driver_set_state(val.val.b);
        } else if(strcmp(esp_rmaker_device_get_name(device),"Switch2")==0) {
            app_driver_set_state2(val.val.b);
        }
        esp_rmaker_param_update_and_report(param, val);
    }
}   
```

adukale, 2021-06-24:
@shahpiyushv Thank you for quick response. 
I am using 4Channel relay module whcih triggers on LOW. So for this I am sending false on receiving true and vice versa.
Modified example code is uploaded here: [](https://github.com/adukale/esp-rainmaker-switch)

I am pretty new to esp-rainmaker and I know I am missing something. Appreciate any kind of help.

Thanks in advance.

adukale, 2021-06-24:
I am not sure if this is issue in my code or a bug: the serial monitor reports correct device name but the  esp_rmaker_param_update_and_report() function prints device name "Switch 1" always. 

       ` esp_rmaker_param_update_and_report(esp_rmaker_device_get_param_by_name(switch2_device, ESP_RMAKER_DEF_POWER_NAME), esp_rmaker_bool(true));`

In this case Switch 2 gets triggered but serial monitor shows Reporting params: {"Switch1":{"Power":true}}

```
I (41955) app_main: Received write request via : Local
I (41955) app_main: Received value = true for Switch2 - Power
I (41975) esp_rmaker_param: Reporting params: {"Switch1":{"Power":true}}
```

shahpiyushv, 2021-06-25:
@adukale , I found what the issue is with your code. You have created a single power parameter and assigned it to all 4 devices

```
    esp_rmaker_param_t *power_param = esp_rmaker_power_param_create(ESP_RMAKER_DEF_POWER_NAME, DEFAULT_POWER);
    esp_rmaker_device_add_param(switch_device, power_param);
    esp_rmaker_device_add_param(switch2_device, power_param);
    esp_rmaker_device_add_param(switch3_device, power_param);
    esp_rmaker_device_add_param(switch4_device, power_param);
```

When you update one or more parameters from the callback, a flag gets set for the parameter indicating that it has updated. Subsequently, a function gets triggered internally that checks the entire device/params database to find all the updated params to be reported and resets that flag. Since switch1 comes first while reading the database, the param gets reported with Switch1 as the device and the flag gets reset.

**You need to create 4 power parameters for your 4 devices.**

As with the true/false logic, it would be better if you manage that in app_driver() (Eg. `gpio_set_level(OUTPUT1_GPIO, !target)`) because that is more as an internal hardware detail. From the user's perspective, a "true" should typically be "ON", irrespective of what the GPIO state needs to be. Ofcourse, since it's your own application, you are free to choose whatever you feel right, but just a suggestion :)

adukale, 2021-06-25:
@shahpiyushv Ahhh! Got it. Fixed the code and it works.
Thank you for the help.