espressif/esp-rainmaker #72: Feature Request: Notification UI Element

JohnMacrae, 2021-05-23:
I believe it is important that we have a UI element to trigger notifications on the Android or IOS device with an appropriate API interface on the ESP32 side. Thank you for providing Rainmaker  - its a great base!

shahpiyushv, 2021-05-24:
@JohnMacrae , notifications is indeed in our near term plan and should be available in a month's time :)

JohnMacrae, 2021-05-24:
Awesome :) Thank you!

warmtek, 2021-06-25:
Thanks for providing us with Rainmaker !!!
Also for us it would be interesting to have the possibility to make notifications on the Android or IOS phone.
As we will need them soon, it is conceivable that they will be implemented soon or we should think about modifying Rainmaker-Android and using a standard parameter to activate them

shahpiyushv, 2021-06-25:
@warmtek significant progress has already been made on cloud side as well as phone app side for various types of notifications. You can expect all of it to converge by mid July.

warmtek, 2021-06-26:
Thank you very much!!!
We will wait the middle of July

warmtek, 2021-09-13:
We are using Rainmaker with notifications and everything is working fine, thanks again.
In an application with Rainmaker we routed Vref2 on IO25 for use in ratiometric temperature measurements with thermistors on ADC1.
Everything is fine until the WIFI starts; at this point Vref2 disconnects from the IO25 which results in high impedance.
Since a printed circuit has been made to use this possibility is it possible to get around this problem ?.

warmtek, 2021-09-13:
We used the master branch on Github fetched just 5 days ago and as explained in "esp_rmaker_core.h" we used the API "esp_rmaker_param_update_and_notify"
We are using ESP-WROOM-32 on DEVKITV1

warmtek, 2021-09-13:
For the routing of VREF2 we used the API that is reported on "adc_common.h" "adc_vref_to_gpio(2,25);"

warmtek, 2021-09-13:
We know that ADC2 is used with WIFI but we thought a static configuration with VREF2 on IO25 was possible.
If this is not possible we will find another solution

JohnMacrae, 2021-09-13:
I think you will have to rework the board. It might be possible to disable
the wifi when taking the reading but that will be very messy with Rainmaker
I think.

Kind Regards
John

shahpiyushv, 2021-09-13:
@warmtek , I think it would be better to ask this question on [esp-idf issues](https://github.com/espressif/esp-idf/issues), since the adc as well as underlying wifi component are part of esp idf. @JohnMacrae support for notifications was added in this commit: https://github.com/espressif/esp-rainmaker/commit/afcc67f3ec4bc4cae1dcfb7d600842bf7da78131. I think that is what you were looking for while filing this issue, so closing this now. Please feel free to write back if you have any further feature requests :)

closed 2021-09-13 by shahpiyushv