espressif/esp-rainmaker #76: homekit_switch  Please set HOMEKIT_PATH to esp-homekit-sdk repo

oksimple, 2021-06-03:
[image]

What is causing this problem?

shahpiyushv, 2021-06-03:
@oksimple Since you are trying to compile the homekit_switch example, you would indeed also require the [esp-homekit-sdk](https://github.com/espressif/esp-homekit-sdk) repo. Have you followed the instructions in this example's [README](https://github.com/espressif/esp-rainmaker/tree/master/examples/homekit_switch)?

oksimple, 2021-06-03:
@shahpiyushv Yes, I strictly follow the documentation to compile, and I can compile several other examples successfully, but the compilation of homekit_switch fails.

shahpiyushv, 2021-06-03:
@oksimple , none of the other examples require the esp-homekit-sdk, so they are expected to compile withoit requiring any additional setup. Have you explicitly followed these steps as per above mentioned [README](https://github.com/espressif/esp-rainmaker/blob/master/examples/homekit_switch/README.md)?

## Setup

Compiling this example would need the [esp-homekit-sdk](https://github.com/espressif/esp-homekit-sdk) repository. Clone it at a suitable location using:

```
git clone --recursive https://github.com/espressif/esp-homekit-sdk.git
```

Export the path to this repository using:

```
export HOMEKIT_PATH=/path/to/esp-homekit-sdk/
```

oksimple, 2021-06-03:
@shahpiyushv  thank you for your answer, I will try it later, and finally, can I use esp-rainmaker commercially?

shahpiyushv, 2021-06-25:
@oksimple , sorry I forgot to reply to your second query. The public RainMaker is not allowed to be used for commercial purposes, but we do have commercial variants of ESP RainMaker. Please send an email to sales@espressif.com for more info.

shahpiyushv, 2021-06-28:
Closing this issue as appropriate answer has been provided.

closed 2021-06-28 by shahpiyushv

jacek12345, 2022-04-02:
Where (from what path) should i start for typing "export HOMEKIT_PATH=/path/to/esp-homekit-sdk/" ? I can only from F:/GitHub/esp-idf. But it seems to not work:
`F:\GitHub\esp-idf>export HOMEKIT_PATH=\GitHub\esp-homekit-sdk
Setting IDF_PATH: F:\GitHub\esp-idf`
When I try from other locations (e.g. from "F:\GitHub\esp-homekit-sdk" i see error that "'export' is not recognized as an internal or external command, operable program or batch file.". 
I got stuck...