espressif/esp-rainmaker #78: Pop contains invalid chars while provisioning in HomeKit Example

adukale, 2021-06-13:
Development Kit: ESP32-DevKitC
Kit version (for WroverKit/PicoKit/DevKitC): DevKitv1
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it):
v4.2
Build System: idf.py
Compiler version (run xtensa-esp32-elf-gcc --version to find it):
8.4.0
Operating System: Windows
(Windows only) environment type: PowerShell
Using an IDE?: No
Power Supply: USB

Pop contains invalid chars while provisioning in HomeKit Example.
Ex.:
https://rainmaker.espressif.com/qrcode.html?data={"ver":"v1","name":"PROV_8BEA1C","pop":"���X57�g","transport":"softap"}
I (1720) app_wifi: Provisioning Started. Name : PROV_8BEA1C, POP : ���X57�g

[image]

adukale, 2021-07-30:
This issue is still there.
Any help is appreciated.
[image]

shahpiyushv, 2021-07-30:
@adukale , we plan to make some more changes to the wifi code in HomeKit, but for the time being, you can use this patch:
[homekit_wifi_patch.txt](https://github.com/espressif/esp-rainmaker/files/6909119/homekit_wifi_patch.txt)

Apply using:

```
git apply homekit_wifi_patch.txt
```

adukale, 2021-07-30:
@shahpiyushv Awesome. This fixed the issue.
The homekit code seems bit complicated to me but considering functionality it is worth diving into.
Cheers...!
Thanks. Appreciate your help.

closed 2021-07-30 by adukale