espressif/esp-rainmaker #81: Use input only pins 36, 39, 34 & 35 as iot_button

adukale, 2021-06-16:
I am designing a circuit to use input only pins 36, 39, 34 & 35 as iot_button. But in the case of 36/39 I get below error in serial monitor once I flash:
```
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57
```

Is there any way to use input only pins as iot_button with iot_button_create or some other way to use these input only GPIOs as buttons using custom declarations?

Thanks

shahpiyushv, 2021-06-16:
@adukale , before we even go further, can you tell me which chip/module/board you are using?

shahpiyushv, 2021-06-17:
@adukale , from the log you shared above, it seems that your flash itself may be corrupted or incorrectly flashed, since the prints do not even show any boot data. Can you double check that first?

adukale, 2021-06-17:
@shahpiyushv I tried with erase_flash but the issue is same.
If I remove reference to pin 39 in iot_button_create then the issue is not occuring. The code works fine in that case.

adukale, 2021-06-22:
@shahpiyushv The pins 36/39 can work as a button in Arduino but not working in Rainmaker.

shahpiyushv, 2021-06-22:
@adukale , I am not sure about Arduino's internal pin to gpio mapping, but do you mean GPIO 36/39 (which are shown as pins 5 and 8) or do you mean PINs 36/39 (which are shown as GPIO 23 and 22)?

adukale, 2021-06-24:
@shahpiyushv I mean GPIO 36/39.
When I declare a variable like this: `#define BUTTON_GPIO          39` then it means GPIO39, correct?

adukale, 2021-07-12:
@shahpiyushv I have added my code here: https://github.com/adukale/esp-rainmaker-switch/tree/8-relay-8-switch

I have added below:
```
#define BUTTON7_GPIO          32
#define BUTTON8_GPIO          33
```
If I add this then idf.py serial monitor gets stuck on `E (1046) gpio: gpio_install_isr_service(438): GPIO isr service already installed`

I also tried with:
```
#define BUTTON7_GPIO          34
#define BUTTON8_GPIO          35
```
For above, idf.py serial monitor doesnot get stuck but the callbacks added doesnot execute.

Thanks in advance.

adukale, 2021-07-12:
> @adukale , this error seems to be coming because of this line:
> 
> https://github.com/espressif/esp-rainmaker/blob/a78fca21bf1e3c15079a3bdbf24a50c55a440a4e/components/button/button/button.c#L247
> 
> Can you try replacing it with the below and see if it fixes the issue?
> 
> ```
> gpio_conf.pin_bit_mask = (uint64_t)1 << gpio_num;
> ```

@shahpiyushv Awesome!!! This fixed the issue. Appreciate your quick response. Thanks.

closed 2021-07-12 by adukale