espressif/esp-rainmaker #82: Claim Verify Failed

Deanzhang0526, 2021-06-17:
The Rainmaker App stopped at Configuring Node association of provisioning, shows "Node addition not confimed"

And the log show below information: 

I (61572) wifi_prov_mgr: Provisioning stopped
E (77672) esp_claim: Invalid response for https://esp-claiming.rainmaker.espressif.com/claim/verify
E (77672) esp_claim: Status = -1, Data = None
E (77672) esp_claim: Claim Verify Failed.
I (77672) app_main: RainMaker Claim Failed.
E (77682) esp_rmaker_core: esp_rmaker_self_claim_perform() returned -1. Aborting
W (77692) esp_rmaker_mqtt: esp_rmaker_mqtt_publish not registered
I (79622) esp_rmaker_time: SNTP Synchronised.
I (79622) esp_rmaker_time: The current time is: Thu Jun 17 17:42:03 2021 +0800[CST], DST: No.

shahpiyushv, 2021-06-17:
@Deanzhang0526 , do you get this error each time? If yes, can you follow the steps mentioned here in the [FAQs](https://rainmaker.espressif.com/docs/faqs.html#how-do-i-fix-the-claim-verify-failed-error-on-esp32-s2)?

shahpiyushv, 2021-11-24:
Closing this issue as it is stale and no response was received. Please update here if the issue still persists.

closed 2021-11-24 by shahpiyushv

Masoud1344, 2022-11-21:
I have this problem too 
What can l solve this problem?

shahpiyushv, 2022-11-21:
@Masoud1344 , the original issue was way back in June 2021. Things may have changed quite a lot since then. Can you file a new issue and mention some more details like chip (esp32c3, esp32s2 or esp32s3?), esp-idf version and logs?