espressif/esp-rainmaker #83: ESP32S2 self claiming verification failed

Shravani100, 2021-07-02:
Hello all,

I am using esp32s2 module on the PCB. I am trying to do self claiming using ./rainmaker.py claim command, But for one of the module it is showing the error as follows :   
Claiming process started. This may take time.
Claim initiate started
Claim initiate done
Generating CSR
[ERROR]:Claim verification failed.
{"status":"failure","description":"User does not have available quota. Please contact Administrator"}

Please guide me for this.

shahpiyushv, 2021-07-02:
@Shravani100 , this error won't come for self claiming, but seems you are doing the host driven claiming. For that, we have a default quota of 5. If you want to increase the quota, just send an email to esp-rainmaker-admin@espressif.com mentioning your registered email address and desired quota.

shahpiyushv, 2021-07-02:
Closing this issue since it has been addressed.

closed 2021-07-02 by shahpiyushv