espressif/esp-rainmaker #84: Problem in executing RAINMAKER SWITCH code on ESP32C3(devKit-M1) through BLUETOOTH

Shravani100, 2021-07-09:
I have executed code of SWITCH (esp-rainmaker--> examples --> switch) from RAINMAKER using ESP32C3. The device was visible on RAINMAKER App and I could switch the LED through WiFi connectivity. But my query is "How to execute same code via ESP32C3's Bluetooth.

shahpiyushv, 2021-07-09:
@Shravani100 , controlling over Bluetooth isn't supported yet.

Shravani100, 2021-07-10:
@shahpiyushv thank you for the information regarding rainmaker and its connectivity to ESP32C3 via bluetooth.
Now with reference **_to access ESP32C3 Bluetooth_**  what I did is, I have downloaded nRF Mesh app to form connectivity between ESP32c3 BLE and mobile. And I could connect device successfully. So is it possible to control (on/off) ESP32C3 via same app using BLE or I need to go for some another way.

shahpiyushv, 2021-11-24:
@Shravani100 , sorry I missed this. This is not really in RainMaker's scope since RainMaker is primarily for cloud communication (with an option of local control over Wi-Fi), but even then could you proceed with what you were trying with BLE mesh?