espressif/esp-rainmaker #87: _eventCallback(): Reason: 2 - AUTH_EXPIRE

stanleyseow, 2021-07-21:
I tried two correct Access Point  with same ERRORS

I cleared the flash and re-flash the RmakerSwitch a few times..

Im using this Wemos R32 ( ESP32 in Uno form factor )
https://www.hackster.io/NYH-workshop/wemos-r32-with-arduino-startup-guide-7bc841

[ 70766][I][RMaker.cpp:19] event_handler(): RainMaker Claim Successful.
[ 95845][W][WiFiGeneric.cpp:829] _eventCallback(): Reason: 2 - AUTH_EXPIRE
E (102590) wifi_prov_mgr: STA Disconnected
E (102590) wifi_prov_mgr: Disconnect reason : 2
E (102594) wifi_prov_mgr: STA Auth Error
[ 95853][E][WiFiGeneric.cpp:448] _arduino_event_cb(): Provisioning Failed: Reason : Authentication Failed
[ 97022][W][WiFiGeneric.cpp:829] _eventCallback(): Reason: 2 - AUTH_EXPIRE
E (103767) wifi_prov_mgr: STA Disconnected
E (103769) wifi_prov_mgr: Disconnect reason : 2
E (103771) wifi_prov_mgr: STA Auth Error
[ 97030][E][WiFiGeneric.cpp:448] _arduino_event_cb(): Provisioning Failed: Reason : Authentication Failed
[ 98149][W][WiFiGeneric.cpp:829] _eventCallback(): Reason: 2 - AUTH_EXPIRE
E (104894) wifi_prov_mgr: STA Disconnected
E (104895) wifi_prov_mgr: Disconnect reason : 2
E (104897) wifi_prov_mgr: STA Auth Error
[ 98156][E][WiFiGeneric.cpp:448] _arduino_event_cb(): Provisioning Failed: Reason : Authentication Failed

shahpiyushv, 2021-07-22:
@stanleyseow , can you add handling for the event ARDUINO_EVENT_PROV_CRED_RECV similar to [this](https://github.com/espressif/arduino-esp32/blob/ee24736042d72dad6e14398d7096a4957325fd3c/libraries/WiFiProv/examples/WiFiProv/WiFiProv.ino#L16), by editing the [sysProvEvent](https://github.com/espressif/arduino-esp32/blob/ee24736042d72dad6e14398d7096a4957325fd3c/libraries/RainMaker/examples/RMakerSwitch/RMakerSwitch.ino#L20) handler and see if you are getting the correct credentials? Can you also once try connecting to an Open AP (no security)?

shahpiyushv, 2021-11-24:
Closing this issue as it is stale and no response was received. Please update here if the issue still persists

closed 2021-11-24 by shahpiyushv

beachmiles, 2022-08-11:
I just got my LOLIN esp32-c3 mini WIFI working by digging this piece of errata from their site. So ANNOYING!!!
```
//You have to add this code after running WiFi.softAP or WiFi.begin for the Wifi to work.
WiFi.setTxPower(WIFI_POWER_8_5dBm);             //https://www.wemos.cc/en/latest/tutorials/c3/get_started_with_arduino_c3.html#wifi
```