espressif/esp-rainmaker #89: PrCan we execute RAINMAKER SWITCH code on ESP32 using WLAN (local network without internet)

ajarsco, 2021-07-28:


shahpiyushv, 2021-07-28:
@ajarsco , the question is not clear to me. Once a device has been added to RainMaker, it can later be controlled on the local network even when there is no Internet. Check the docs [here](https://rainmaker.espressif.com/docs/local-control.html) for more info.

ajarsco, 2021-07-28:
> @ajarsco , the question is not clear to me. Once a device has been added to RainMaker, it can later be controlled on the local network even when there is no Internet. Check the docs [here](https://rainmaker.espressif.com/docs/local-control.html) for more info.

Ok thank you sir, let me check

shahpiyushv, 2021-08-11:
Closing this issue since it has been addressed. Do let us know if any additional information is required.

closed 2021-08-11 by shahpiyushv